loboris/MicroPython_ESP32_psRAM_LoBo #77: Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

wootsrx, 2018-02-23:
**Dear Loboris**

**Sorry for my poor English.**
**I would like to get your help, i faced trouble with the following error,** 
**Could you please suggest me about the error cause?** 
**I don't want to put my long code here so I 'm trying to minimize the code which can reproduce the error** **but not finish yet.**
**PS. The error happened after I put my code as main.py , if I just run on python prompt it was not appear.**

**Thank you in advance for your kindly help.**

Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x401185ec  PS      : 0x00060330  A0      : 0x80121355  A1      : 0x3ffb7ee0
0x401185ec: qstr_find_strn at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/qstr.c:173

A2      : 0x0fd032b2  A3      : 0x00000000  A4      : 0x3ffb2f20  A5      : 0x3ffc9050
A6      : 0x00000001  A7      : 0x3ffc0b90  A8      : 0x801185ec  A9      : 0x3ffb7ec0
A10     : 0x00000081  A11     : 0x00000000  A12     : 0x000001de  A13     : 0x3ffc0890
A14     : 0x00000203  A15     : 0x00000460  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000081  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x401185ec:0x3ffb7ee0 0x40121352:0x3ffb7f00 0x401214e6:0x3ffb7f30 0x40121463:0x3ffb7f50 0x401234e9:0x3ffb7f80 0x4011fc41:0x3ffb7fa0 0x4012bf83:0x3ffb7fc0 0x40124
1aa:0x3ffb8060 0x4011fde5:0x3ffb8090 0x4011fe12:0x3ffb80b0 0x401136e5:0x3ffb80d0 0x4011396d:0x3ffb8180 0x4010e33a:0x3ffb81a0
0x401185ec: qstr_find_strn at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/qstr.c:173

0x40121352: mp_map_init at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/map.c:75

0x401214e6: mp_map_lookup at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/map.c:190

0x40121463: mp_map_lookup at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/map.c:172

0x401234e9: dict_print at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objdict.c:61

0x4011fc41: mp_make_function_from_raw_code at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/emitglue.c:144 (discriminator 1)

0x4012bf83: mp_execute_bytecode at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:256

0x401241aa: fun_builtin_0_call at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:51

0x4011fde5: mp_store_name at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x4011fe12: mp_delete_name at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x401136e5: parse_compile_execute at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:531

0x4011396d: pyexec_friendly_repl at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:492

0x4010e33a: mp_task at C:/msys32/home/USER/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/main.c:147

CPU halted.

loboris, 2018-03-06:
Do you still have the issue.
Could you post at least part of your main.py which causes the error ?

wootsrx, 2018-03-07:
Dear Loboris,

Thank you very much for your reply.
I am not sure my code can reproduce the error at your site or not because it happened when I communicate with external device by UART (RS485).
Also I put a module as frozen module (put in modules directory before make).

Can I just send you the module file and main.py ? Sorry my ability is so limited.

Thank you so much for your kindly help.

Woot

loboris, 2018-03-07:
Yes, please, if possible, send me the module and the main.py.

wootsrx, 2018-03-10:
So sorry for my delay, I just send files to your e-mail. 
Thank you very much for your kindly help.

loboris, 2018-04-12:
Several updates were commited after this issue was opened.

Can you confirm you still have the issues after updating to the latest commit.

lypanov, 2018-05-01:
I also see the same error though under very different circumstances. Should I create a new issue or add a reduction here?

lypanov, 2018-05-02:
My issue is fixed (oddly...) by removing all print() debugging statements from my MicroWebSrv handlers.

loboris, 2018-05-17:
The issues should be resolved after the latest updates.
Please reopen if needed.

closed 2018-05-17 by loboris

robertelhelo, 2019-02-28:
Hi @loboris ,
 I would like to reopen this issue as the problem still persists. I experience it sporadically when I close a websocket connection to Microwebserver. It happens consistently when I create two client ws connections and close any one of them.

 Attaching my main class containing websocket handlers.

[basestationgps.zip](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/2914225/basestationgps.zip)

The Guru Mediation error is the following:
GPS read thread started.
Clients before: []
Clients after: [<MicroWebSocket object at 3ffcfb70>]
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x401d098f  PS      : 0x00060330  A0      : 0x800fedf3  A1      : 0x3fff6770
A2      : 0x400f71c1  A3      : 0x3ffca290  A4      : 0x000000ff  A5      : 0x0000007f
A6      : 0x00000082  A7      : 0x00000004  A8      : 0x3f4190d8  A9      : 0x400f71c0
A10     : 0x00000000  A11     : 0x3f41883c  A12     : 0x3ffd6110  A13     : 0x00000000
A14     : 0xfffffffe  A15     : 0x3ffd6120  SAR     : 0x0000001e  EXCCAUSE: 0x00000003
EXCVADDR: 0x400f71c0  LBEG    : 0x4009bb2c  LEND    : 0x4009bb5a  LCOUNT  : 0xffffffff

Backtrace: 0x401d098f:0x3fff6770 0x400fedf0:0x3fff6790 0x400f527e:0x3fff6830 0x400f04b9:0x3fff6860 0x400f0529:0x3fff6880 0x400fe83e:0x3fff68a0 0x400f527e:0x3fff6940 0x400f04b9:0x3fff6970 0x400f3550:0x3fff6990 0x400f356d:0x3fff69e0 0x400f04b9:0x3fff6a00 0x400fd019:0x3fff6a20 0x400dc6b5:0x3fff6ac0

CPU halted.