loboris/MicroPython_ESP32_psRAM_LoBo #78: ftp CPU halted

kevinkk525, 2018-02-24:
I wanted to use ftp module to sync some files but no matter what client I use, the esp panic'ed when executing any command. sometimes listing the files works once but crashes on the second try.

Error message:
```
Guru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x40172e43  PS      : 0x00060630  A0      : 0x801717a9  A1      : 0x3ffcf3d0
0x40172e43: SPIFFS_readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1125

A2      : 0x3ffd09a0  A3      : 0x00000000  A4      : 0x00060f20  A5      : 0x00000001
A6      : 0x00060123  A7      : 0x3ffcf460  A8      : 0x80172e10  A9      : 0x3ffcf390
A10     : 0xffffd8a8  A11     : 0x00000260  A12     : 0x00000260  A13     : 0x0000154f
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000044  LBEG    : 0x4009e01c  LEND    : 0x4009e04a  LCOUNT  : 0x00000000
0x4009e01c: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x4009e04a: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

Backtrace: 0x40172e43:0x3ffcf3d0 0x401717a6:0x3ffcf410 0x4017193c:0x3ffcf4e0 0x40178529:0x3ffcf510 0x4011d723:0x3ffcf530 0x4011e5a9:0x3ffcf550 0x40115a9f:0x3ffcf580
0x40172e43: SPIFFS_readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1125

0x401717a6: vfs_spiffs_readdir_r at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:434

0x4017193c: vfs_spiffs_readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:434

0x40178529: readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/vfs/./vfs.c:464

0x4011d723: ftp_list_dir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c:1326

0x4011e5a9: ftp_run at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c:1326

0x40115a9f: ftp_task at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/mpthreadport.c:512

CPU halted.
```

(sysname='esp32', nodename='esp32', release='3.1.18', version='ESP32_LoBo_v3.1.18 on 2017-02-07', machine='ESP32 board with ESP32')

loboris, 2018-03-06:
Do you still have this issue?
Can you give more information, have you built the firmware or are using the prebuilt firmware, did you change something in the configuration, are you using psRAM, ...

kevinkk525, 2018-03-06:
yes I still have the issue after the connection when I want to get the files inside "flash":

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40136300  PS      : 0x00060e30  A0      : 0x801374f1  A1      : 0x3ffcf600
0x40136300: spiffs_api_unlock at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:453

A2      : 0x3ffd0a9c  A3      : 0x00000260  A4      : 0x00000260  A5      : 0x0000154f
A6      : 0x00000000  A7      : 0x00000004  A8      : 0xe8075eb2  A9      : 0x00000001
A10     : 0x0000000f  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x00000100  A15     : 0x3ffcf660  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0xe8075eb6  LBEG    : 0x400990e4  LEND    : 0x40099112  LCOUNT  : 0x00000000
0x400990e4: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x40099112: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

Backtrace: 0x40136300:0x3ffcf600 0x401374ee:0x3ffcf620 0x40135e0d:0x3ffcf660 0x40135fac:0x3ffcf730 0x4013cbc1:0x3ffcf760 0x400dfc23:0x3ffcf780 0x400e0a61:0x3ffcf7a0 0x400d8ec2:0x3ffcf7d0
0x40136300: spiffs_api_unlock at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:453

0x401374ee: SPIFFS_readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1125

0x40135e0d: vfs_spiffs_readdir_r at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:453

0x40135fac: vfs_spiffs_readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/spiffs/./esp_spiffs.c:453

0x4013cbc1: readdir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/vfs/./vfs.c:464

0x400dfc23: ftp_list_dir at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c:1329

0x400e0a61: ftp_run at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/libs/ftp.c:1329

0x400d8ec2: ftp_task at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/mpthreadport.c:506

CPU halted.
```

Version: (sysname='esp32_LoBo', nodename='esp32_LoBo', release='3.1.21', version='ESP32_LoBo_v3.1.21 on 2017-03-03', machine='ESP32 board with ESP32')

firmware is manually built with psram, sdkconfig:

```
#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=y
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=2
CONFIG_BOOTLOADER_VDDSDIO_BOOST=y

#
# Security features
#
CONFIG_SECURE_BOOT_ENABLED=
CONFIG_FLASH_ENCRYPTION_ENABLED=

#
# Serial flasher config
[... 598 lines cut ...]
#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE_4096=
CONFIG_WL_SECTOR_SIZE=512
CONFIG_WL_SECTOR_MODE_PERF=
CONFIG_WL_SECTOR_MODE_SAFE=y
CONFIG_WL_SECTOR_MODE=1
```

kevinkk525, 2018-03-07:
I just tried again without a main.py that was starting my asyncio loop. Now the ftp does not crash. I can see my files but download does not work (using filezilla configured according to wiki: Antwort:	150 
Antwort:	451, Error). And on an upload the same. Sometimes it works but still throws an error.

So it seems that the ftp module crashes if I had an asyncio loop running.

loboris, 2018-03-07:
I've been using ftp for many months and never had a crash.
But it was not tested with asyncio.

I do not recommend using asyncio with this port. You can use threads to achieve the same (or, IMHO, better) functionality.

kevinkk525, 2018-03-07:
In my understanding asyncio and ftp should not have a problem with each other as they run in separate threads.

Well this debate is already going on for quite some time and IMHO using asyncio is the better option as I have full control over when I release the control to a scheduler. Apart from that I have been writing a lot of stuff using asyncio already, so I'll not switch to threads, especially as this stuff is supposed to run on an esp8266 as well (ran there first actually).

But as you do not recommend using asyncio with your port, I'd understand if you do not spend time on solving the problem between asyncio and the ftp module (or thread implementation).

loboris, 2018-03-07:
Could you send some simplest example which produces this crash ?
I would like to try to make it work.
The problem may be caused by the fact that the main MicroPython task (as FreeRTOS task) and the ftp task runs with the same priority.

kevinkk525, 2018-03-07:
Funny enough it turns out that it is not even related to asyncio at all:

```
import time
import network
import machine
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
wlan.connect("SSID","PASS")
while wlan.isconnected() == False:
    time.sleep(0.5)
    machine.idle()
    print("not connected")
print("connected")
network.ftp.start()
```

Once this is done I connected using filezilla. If I want to get the contents of the directory "flash" it crashes.
This does not occur if I start the ftp module manually after the main.py connected.

loboris, 2018-03-07:
Here is the complete **main.py** I'm using:
```python
import machine, network, utime,

print("")
print("Starting WiFi ...")

sta_if = network.WLAN(network.STA_IF)

_ = sta_if.active(True)
_ = sta_if.connect("my_SSID", "my_PASS")

tmo = 50
while not sta_if.isconnected():
    utime.sleep_ms(100)
    tmo -= 1
    if tmo == 0:
        break

if tmo > 0:
    ifcfg = sta_if.ifconfig()
    print("WiFi started, IP:", ifcfg[0])
    utime.sleep_ms(500)

    rtc = machine.RTC()
    print("Synchronize time from NTP server ...")
    rtc.ntp_sync(server="hr.pool.ntp.org") # update once
    #rtc.ntp_sync(server="hr.pool.ntp.org", update_period=3600)
    tmo = 100
    while not rtc.synced():
        utime.sleep_ms(100)
        tmo -= 1
        if tmo == 0:
            break

    if tmo > 0:
        utime.sleep_ms(200)
        print("Time set:", utime.strftime("%c"))
        print("")
        _ = network.ftp.start()
        _ = network.telnet.start()
        try:
            mdns = network.mDNS()
            _ = mdns.start("mPy","MicroPython with mDNS")
            _ = mdns.addService('_ftp', '_tcp', 21, "MicroPython", {"board": "ESP32", "service": "mPy FTP File transfer", "passive": "True"})
            _ = mdns.addService('_telnet', '_tcp', 23, "MicroPython", {"board": "ESP32", "service": "mPy Telnet REPL"})
            _ = mdns.addService('_http', '_tcp', 80, "MicroPython", {"board": "ESP32", "service": "mPy Web server"})
        except:
            print("mDNS not available")
```
After it finishes (and wifi is connected and ftp started) I can connect with Filezilla and list, download, upload, without any problems/crashes.

BDW, `machine.idle()` is not needed, `time.sleep()` does not block the other tasks.

kevinkk525, 2018-03-07:
Interesting, that works. But a little modifications were enough to get it struggling again. Important seems also to have filezilla closed and start a fresh connection.

```
import machine, network, utime

print("")
print("Starting WiFi ...")

sta_if = network.WLAN(network.STA_IF)

_ = sta_if.active(True)
_ = sta_if.connect("my_SSID", "my_PASS")

tmo = 50
while not sta_if.isconnected():
    utime.sleep_ms(100)
    tmo -= 1
    if tmo == 0:
        break

if tmo > 0:
    ifcfg = sta_if.ifconfig()
    print("WiFi started, IP:", ifcfg[0])
    utime.sleep_ms(500)

    if tmo > 0:
        utime.sleep_ms(200)
        print("Time set:", utime.strftime("%c"))
        print("")
        _ = network.ftp.start()
        #_ = network.telnet.start()
        try:
            #mdns = network.mDNS()
            #_ = mdns.start("mPy","MicroPython with mDNS")
            #_ = mdns.addService('_ftp', '_tcp', 21, "MicroPython", {"board": "ESP32", "service": "mPy FTP File transfer", "passive": "True"})
            pass
        except:
            print("mDNS not available")
```

With mdns enabled it seems to work reliable.

loboris, 2018-03-18:
@kevinkk525

The ftp server was updated recently, do you still have issues with it ?

kevinkk525, 2018-03-18:
I retried it a few times with a minimal script and did not encounter a problem so I'll close this issue.

closed 2018-03-18 by kevinkk525