loboris/MicroPython_ESP32_psRAM_LoBo #80: TFT support on ESP-WROVER-KIT not working

nickovs, 2018-03-01:
The TFT display driver using the [example](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/display#example) settings in the Wiki documentation listed for the 'ESP32-WROVER-KIT v3' do not appear to work. The LCD on our unit preforms the demo/test pattern display with the default firmware that Espressif ships but using the example settings we are unable to get anything to draw. The backlight functions correctly and it would appear that the assignment of pins in the example matches the tracks on the PCB but we are not seeing any signs of life at all from the display when running MicroPython.

Has this code been tested on the WROVER kit or is it written from the specification? If there is some known-to-be-good test code I'd happily run it on the hardware I have to check. If it's not been tested then we can look into getting a test device to the author if there is interest.

nickovs, 2018-03-01:
FYI, the tests were performed with the psRAM, all modules [prebuilt firmware](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/tree/42bd0735031c8c73c591e0bcda2817444ab65a89/MicroPython_BUILD/firmware/esp32_psram_all) in the 42bd073 commit.

loboris, 2018-03-01:
I've flashed the same prebuilt firmware on my ESP-WROVER-KIT-v3 and the display works fine.
I'm not shure if all WROVER-KIT v3 have the same ST7789 display.

nickovs, 2018-03-01:
Thanks for testing this. Are you just using the wiki example with the appropriate `init()` line uncommented or did you use a different test? If it's a different test would you mind copy/pasting the test code to make sure that I'm not doing something dumb? Otherwise I will try to work out if my display is a different model (which might involve unsticking it from the board).

loboris, 2018-03-01:
After boot:
```
import display
tft = display.TFT()
# ESP32-WROVER-KIT v3:
tft.init(tft.ST7789, rst_pin=18, backl_pin=5, miso=25, mosi=23, clk=19, cs=22, dc=21)
```
initializes the display and 'MicroPython' string in Red, Gree, and Blue is printed in the middle of the display.

All other functions works too.

nickovs, 2018-03-01:
Thanks. For me that just brings the backlight on but does nothing else. I'll see if I can take it apart and find out if there is a different controller on the card.

nickovs, 2018-03-01:
OK. It seems that the unit that I have has an ILI9341 on it. The following init line worked for my ESP-WROVER-KIT:
```
tft.init(tft.ILI9341, rst_pin=18, backl_pin=5, miso=25, mosi=23, clk=19, cs=22, dc=21)
```
I get the RGB MicroPython message and drawing commands work.

It might make sense to update the Wiki.

closed 2018-03-02 by nickovs

ronanmcall, 2018-04-30:
I'm probably doing something wrong...

I have a problem on my new ESP32-WROVER-KIT v3 -- I was able to get the Expressif LCD test C code working fine forcing the display to be of type IL9341,  but not getting it to work using the MicroPython Example OOTB.

Nick, if you are online, do you have any jumpers set on your board JP13?  I get the following issue:

>>> import display
>>> tft = display.TFT()
>>> # ESP32-WROVER-KIT v3:
>>> tft.init(tft.ST7789, rst_pin=18, backl_pin=5, miso=25, mosi=23, clk=19, cs=22, dc=21)
E (2182184) spi_master: spi_bus_add_device(234): no free cs pins for host
E (2182185) [SPI_UTILS]: Adding spi device failed with rc=0x105
E (2182188) spi_master: spi_bus_free(178): not all CSses freed

Same thing if I substitute ILI9341 above -- and BTW, the Expressif LCD test code/demo in C works if I use the following configuration (with help from Lobo):

----------- works using Expressif C LCD test code ---------------

**Lobo>**"change the controller type in main/tft_demo.c, comment tft_disp_type = DEFAULT_DISP_TYPE; and uncomment tft_disp_type = DISP_TYPE_ILI9341; and build again."

In Make menuconfig I did set the ESP32-WROVER-KIT display option but after that I set the below in ~/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/IL9341-test/ESP32_TFT_library/components/tft/tftspi.h:

// ** Set the correct configuration for ESP32-WROVER-KIT v3
// --------------------------------------------------------
#define DEFAULT_DISP_TYPE           DISP_TYPE_ST7789V **<-- overridden in code to be ILO9341**
#define DEFAULT_TFT_DISPLAY_WIDTH   240
#define DEFAULT_TFT_DISPLAY_HEIGHT  320
#define DISP_COLOR_BITS_24          0x66
#define DEFAULT_GAMMA_CURVE         0
#define DEFAULT_SPI_CLOCK           26000000
#define TFT_INVERT_ROTATION         0
#define TFT_INVERT_ROTATION1        0
// RONAN -- use below TFT_RGB_BGR 0x08
#define TFT_RGB_BGR                 0x08
//#define TFT_RGB_BGR                 0x00

fivdi, 2018-07-01:
The content of this issue was very helpful. The ESP32-WROVER-KIT that I have also appears to have an ILI9341. Calling `tft.init` with the display controller type set to `tft.ILI9341` resolved the issue for me too.

@ronanmcall No jumpers were used on the JP13 jumper pins.

BhuvanKanade, 2018-09-06:
HI,
@nickovs 
where have you connected backl_pin pin ?
Can you share schematic or pin connection of it ?
Hav eyou used any circuit to make level shifter etc ?

Thank you in advance.

nickovs, 2018-09-06:
@BhuvanKanade I was using the pre-built [ESP32-WROVER-KIT](https://www.adafruit.com/product/3384) which has the display built on board. I have subsequently used an external ILI9341 and this has not needed any level shifters; I just connected the signals to pins that were not otherwise in use. I hope this helps.