loboris/MicroPython_ESP32_psRAM_LoBo #81: [Mqtt client]: Failed to create socket. Connection to server mqtt.thingspeak.com:1883 failed!

stroobandt, 2018-03-01:
First, some context: I have ample experience with successfully connecting to the popular, freemium [ThingSpeak](https://thingspeak.com/) service using the standard MicroPython `umqtt` module.

I have carefully studied [your `mqtt` interface](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/mqtt) as well as the [ThingSpeak MQTT API](https://nl.mathworks.com/help/thingspeak/mqtt-basics.html) and especially the [ThingSpeak MQTT pitfalls](https://nl.mathworks.com/help/thingspeak/troubleshoot-MQTT-publish.html).

Despite all this background, I fail to successfully connect to the ThingSpeak MQTT server.
Using `mqtt.status()` and the REPL, I witness several connection attempts. Eventually, the following errors are printed in the REPL (the number between the parenthesis varies):

    E (1786998) [Mqtt client]: Failed to create socket
    E (1786999) [Mqtt client]: Connection to server mqtt.thingspeak.com:1883 failed!

The `connected_cb` callback function gets never called.
With respect to the [ThingSpeak MQTT troubleshooting page](https://nl.mathworks.com/help/thingspeak/troubleshoot-MQTT-publish.html), I have tried many variations of `autoreconnect=True` and `cleansession=True`, etc.

In view of all this, I am starting to suspect a bug. I would love to see a successful ThingSpeak MQTT example in the documentation.

carterw, 2018-03-01:
Failing to create the socket sounds unrelated to the MQTT protocol. Do you have a WiFi connection to the internet actively up and running while you are attempting to reach the MQTT server?

stroobandt, 2018-03-02:
@carterw Yes, certainly, and I always test for it with `wlan.isconnected()`.
I really do think it must have something to do with Loboris' implementation of `mqtt`.
With standard `umqtt` and renaming the keywords, I have no issues.

stroobandt, 2018-03-28:
@carterw There are two ways to convey MQTT messages, either over TCP or websockets.

Anyhow, I am going to close this issue now as Thingspeak admitted in a private message that they have issues with MQTT server reliability.
However, there are no reliability issues when using [Thingspeak's REST API](https://www.mathworks.com/help/thingspeak/channels-and-charts-api.html).

closed 2018-03-28 by stroobandt

Josverl, 2018-04-11:
I am running into the same. I have a few network traces that indicate that the issue is caused by the LoBo built-in MQTT client requiring a \lwt which is not allowed by ThingSpeak 

Lobo Built-in client 
Frame 256: 116 bytes on wire (928 bits), 116 bytes captured (928 bits) on interface 0
Ethernet II, Src: 62:ea:17:ea:c2:9d (62:ea:17:ea:c2:9d), Dst: IETF-VRRP-VRID_02 (00:00:5e:00:01:02)
Internet Protocol Version 4, Src: 10.137.249.246, Dst: 52.54.126.159
Transmission Control Protocol, Src Port: 62310, Dst Port: 1883, Seq: 1, Ack: 1, Len: 62
MQ Telemetry Transport Protocol, Connect Command
    Header Flags: 0x10 (Connect Command)
    Msg Len: 60
    Protocol Name Length: 6
    Protocol Name: MQIsdp
    Version: MQTT v3.1 (3)
    Connect Flags: 0xc4
        1... .... = User Name Flag: Set
        .1.. .... = Password Flag: Set
        ..0. .... = Will Retain: Not set
        ...0 0... = QoS Level: At most once delivery (Fire and Forget) (0)
        .... .1.. = Will Flag: Set
        .... ..0. = Clean Session Flag: Not set
        .... ...0 = (Reserved): Not set
    Keep Alive: 120
    Client ID Length: 15
    Client ID: mpy_mqtt_client
    Will Topic Length: 4
    Will Topic: /lwt
    Will Message Length: 0
    Will Message: 
    User Name Length: 3
    User Name: jos
    Password Length: 16
    Password: U31KDACARRXDWGSF

Working Desktop client 
Frame 4: 101 bytes on wire (808 bits), 101 bytes captured (808 bits) on interface 0
Ethernet II, Src: 6a:45:33:f3:f8:38 (6a:45:33:f3:f8:38), Dst: Fortinet_09:1f:0c (00:09:0f:09:1f:0c)
Internet Protocol Version 4, Src: 172.17.1.67, Dst: 34.235.10.220
Transmission Control Protocol, Src Port: 50463, Dst Port: 1883, Seq: 1, Ack: 1, Len: 47
MQ Telemetry Transport Protocol, Connect Command
    Header Flags: 0x10 (Connect Command)
    Msg Len: 45
    Protocol Name Length: 4
    Protocol Name: MQTT
    Version: MQTT v3.1.1 (4)
    Connect Flags: 0xc2
        1... .... = User Name Flag: Set
        .1.. .... = Password Flag: Set
        ..0. .... = Will Retain: Not set
        ...0 0... = QoS Level: At most once delivery (Fire and Forget) (0)
        .... .0.. = Will Flag: Not set
        .... ..1. = Clean Session Flag: Set
        .... ...0 = (Reserved): Not set
    Keep Alive: 60
    Client ID Length: 10
    Client ID: client_123
    User Name Length: 3
    User Name: jos
    Password Length: 16
    Password: U32KDACARRXDWGSF

Non-Working

stroobandt, 2018-04-12:
@Josverl Feel free to reopen this issue.
Mind you, it is not a good idea to post passwords over here!

Josverl, 2018-04-13:
Well , the password/APIKey shown is **_not_** the real APIKey :-) 
And perhaps its my oversight , but I cannot see an option to re-open, only to create a new issue.
Ill do that and ref this one then.