loboris/MicroPython_ESP32_psRAM_LoBo #82: Can't sync local time correctly.

djairjr, 2018-03-01:
Boris, I'm in São Paulo, Brazil.
You are trying to synchronize the clock using the rtc.ntp_sync (server [, update_period] [, tz]) method and setting tz = <- 03> 3 <-02>, according to line 180 of the zones.csv file (America_Sao_Paulo settings)

```
rtc = machine.RTC ()
rtc.ntp_sync (server = "a.ntp.br", tz = "<-03> 3 <-02>")
tmo = 100
     while not rtc.synced ():
         utime.sleep_ms (100)
         tmo - = 1
         if tmo == 0:
             break

```

It turns out that the time displayed is not correct. It's like 6 hours more than here in Brazil.

loboris, 2018-03-06:
Have you tried to include the whole string `"<-03>3<-02>,M10.3.0/0,M2.3.0/0"` ?
Also there should be no spaces in the tz string.

djairjr, 2018-03-07:
Ok. I will try it.

djairjr, 2018-03-07:
Ok, Boris, I did two tests here and none of them were satisfactory. 

**First** I changed the timezone string in my boot.py file, according to your suggestion. The clock still remained 6 hours in advance.

_(But I had also made the same timezone change in the firmware, using the option that appears in the menu when I run ./BUILD.sh menuconfig)_

**Second**,  I re-done the whole process by changing the timezone only in the firmware, without doing the same setting in my boot.py.
This time the clock was only 3 hours in advance. 

But it still does not show the correct time.

loboris, 2018-03-08:
Please, see the replay on the [forum](https://loboris.eu/forum/showthread.php?tid=12&pid=21#pid21) for explanation and solution.

loboris, 2018-03-24:
As getting the correct time from NTP server depends on setting the correct time zone string, and otherwise works correctly, I will cloce this issue.
Please reopen if needed.

closed 2018-03-24 by loboris