loboris/MicroPython_ESP32_psRAM_LoBo #84: Build Failing when using SD SPI Mode

neoxharsh, 2018-03-02:
Build if failing if I change the sd card mode to SPI and give in the pin number.

```
In file included from D:/MPL/MicroPython_BUILD/components/micropython/esp32/modmachine.c:47:0:
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modmachine.c: At top level:
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modmachine.c:544:65: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(mod_machine_stdin_disable_obj, mod_machine_stdin_disable);
                                                                 ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:285:46: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_0'
         {{&mp_type_fun_builtin_0}, .fun._0 = fun_name}
                                              ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modmachine.c:544:65: note: (near initialization for 'mod_machine_stdin_disable_obj.fun._0')
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(mod_machine_stdin_disable_obj, mod_machine_stdin_disable);
                                                                 ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:285:46: note: in definition of macro 'MP_DEFINE_CONST_FUN_OBJ_0'
         {{&mp_type_fun_builtin_0}, .fun._0 = fun_name}
                                              ^
_CC_ /d/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c: In function 'processEvent_callback':
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:298:32: warning: unused variable 'info' [-Wunused-variable]
     system_event_sta_got_ip_t *info = (system_event_sta_got_ip_t *)&event->event_info;
                                ^
In file included from D:/MPL/MicroPython_BUILD/components/micropython/py/mpstate.h:36:0,
                 from D:/MPL/MicroPython_BUILD/components/micropython/py/nlr.h:68,
                 from D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:43:
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c: At top level:
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:937:36: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_start),  MP_ROM_PTR(&mod_network_startTelnet_obj) },
                                    ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:938:36: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_pause),  MP_ROM_PTR(&mod_network_pauseTelnet_obj) },
                                    ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:939:36: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_resume), MP_ROM_PTR(&mod_network_resumeTelnet_obj) },
                                    ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:940:34: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_stop), MP_ROM_PTR(&mod_network_stopTelnet_obj) },
                                  ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:941:36: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_status), MP_ROM_PTR(&mod_network_stateTelnet_obj) },
                                    ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:942:35: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_stack), MP_ROM_PTR(&mod_network_TelnetMaxStack_obj) }
                                   ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
                       ^
D:/MPL/MicroPython_BUILD/components/micropython/esp32/modnetwork.c:1080:36: note: in expansion of macro 'MP_ROM_PTR'
     { MP_ROM_QSTR(MP_QSTR_start),  MP_ROM_PTR(&mod_network_startFtp_obj) },
                                    ^
D:/MPL/MicroPython_BUILD/components/micropython/py/obj.h:242:23: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 #define MP_ROM_PTR(p) (p)
[... 189 lines cut ...]
  SMS_indexes indexes;
              ^
_CC_ /d/MPL/MicroPython_BUILD/components/micropython/esp32/network_mdns.c
_CC_ /d/MPL/MicroPython_BUILD/components/micropython/extmod/modbtree.c
In file included from D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:36:0,
                 from D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/btree/btree.h:44,
                 from D:/MPL/MicroPython_BUILD/components/micropython/extmod/modbtree.c:38:
D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/PORT/include/bsd-queue.h:1:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 queue.h
      ^
In file included from D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/btree/btree.h:44:0,
                 from D:/MPL/MicroPython_BUILD/components/micropython/extmod/modbtree.c:38:
D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:59:3: warning: data definition has no type or storage class
 } BKT;
   ^
D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:59:3: error: type defaults to 'int' in declaration of 'BKT' [-Werror=implicit-int]
D:/MPL/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:62:2: error: expected specifier-qualifier-list before 'CIRCLEQ_HEAD'
  CIRCLEQ_HEAD(_lqh, _bkt) lqh; /* lru queue head */
  ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/d/MPL/MicroPython_BUILD/components/micropython/py/mkrules.mk:48: /d/MPL/MicroPython_BUILD/build/extmod/modbtree.o] Error 1
make: *** [/d/MPL/Tools/esp-idf/make/project.mk:451: component-micropython-build] Error 2
'make all' FAILED!

```

loboris, 2018-03-06:
After you copy the _esp32_psram_all_ sdkconfig to _sdkconfig_ run `./BUILD.sh menuconfig` first, then `./BUILD.sh clean`.
After that the build process should finish without errors.

loboris, 2018-03-18:
I'll close the issue. Please reopen if needed.

closed 2018-03-18 by loboris

drschlaumeier, 2019-03-23:
Hallo,
I did all the suggestions above but build still fails. 
1. copy esp32_psram_all sdkconfig to sdkconfig in BUILD.sh directory
2. ./BUILD.sh menuconfig
3. ./BUILD.sh clean
4. ./BUILD.sh all -v
What is wrong? Any more suggestions?

In file included from D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:36:0,
                 from D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/btree/btree.h:44,
                 from D:/data2/micropyton/MicroPython_BUILD/components/micropython/extmod/modbtree.c:39:
D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/PORT/include/bsd-queue.h:1:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 queue.h
      ^
In file included from D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/btree/btree.h:44:0,
                 from D:/data2/micropyton/MicroPython_BUILD/components/micropython/extmod/modbtree.c:39:
D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:59:3: warning: data definition has no type or storage class
 } BKT;
   ^
D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:59:3: error: type defaults to 'int' in declaration of 'BKT' [-Werror=implicit-int]
D:/data2/micropyton/MicroPython_BUILD/components/micropython/lib/berkeley-db-1.xx/include/mpool.h:62:2: error: expected specifier-qualifier-list before 'CIRCLEQ_HEAD'
  CIRCLEQ_HEAD(_lqh, _bkt) lqh; /* lru queue head */
  ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/d/data2/micropyton/MicroPython_BUILD/components/micropython/py/mkrules.mk:48: /d/data2/micropyton/MicroPython_BUILD/build/extmod/modbtree.o] Fehler 1
make: *** [/d/data2/micropyton/Tools/esp-idf/make/project.mk:468: component-micropython-build] Fehler 2
'make all' FAILED!

drschlaumeier, 2019-03-24:
Ahh, OK, found the problem :-)
I'm compiling under Win10 & mingw32. Seems that compiler cc1.exe does not understand the include in file bsd-queue.h. Renaming _queue.h_ to _#include "queue.h"_ solved the issue.
BR