loboris/MicroPython_ESP32_psRAM_LoBo #87: build errors from fresh clone

carterw, 2018-03-06:
Performed a fresh clone, went into build menu and turned off telnet and FTP here;
menu->MicroPython->System settings

./BUILD.sh all -v

/micropython/py/modthread.c: In function 'mod_thread_list':
/micropython/py/modthread.c:450:24: error: 'TELNET_STACK_LEN' undeclared (first use in this function)
      TelnetTaskHandle, TELNET_STACK_LEN, TELNET_STACK_LEN - uxTaskGetStackHighW/micropython/py/modthread.c:454:21: error: 'FTP_STACK_LEN' undeclared (first use in this function)
      FtpTaskHandle, FTP_STACK_LEN, FTP_STACK_LEN - uxTaskGetStackHighWaterMark(

Turned telnet and FTP back on, ran clean and rebuilt successfully.

ldecicco, 2018-03-06:
I confirm the issue.

loboris, 2018-03-06:
Thanks for reporting this.
The fix is commited and building without telnet and/or ftp server enabled works.

carterw, 2018-03-07:
Seems to be working now, thanks.

closed 2018-03-07 by carterw