loboris/MicroPython_ESP32_psRAM_LoBo #88: build error with power management

MZachmann, 2018-03-06:
I tried to build with power management enabled and not too much else and got the following error:

/mnt/c/microbld/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/driver/include/driver/spi_master_internal.h:55:5: **error: unknown type name 'esp_pm_lock_handle_t'
     esp_pm_lock_handle_t pm_lock;**
     ^
In file included from /mnt/c/microbld/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/driver/include/driver/spi_master_utils.h:20:0,
                 from /mnt/c/microbld/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/driver/include/driver/sdspi_host.h:24,
                 from /mnt/c/microbld/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/esp_vfs_fat.h:21,
                 from /mnt/c/microbld/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/vfs_fat_spiflash.c:19:

I added the include line:

**#include "esp_pm.h"**

to _spi_master_internal.h_

loboris, 2018-03-06:
Power management is not implemented in MicroPython at the moment.

The feature is planned, when it will be implemented there will be an option im MicroPython configuration (in menuconfig) to enable it.

MZachmann, 2018-03-06:
I had options in the menuconfig for it and turned on only the top one (enable). It did build (with the include change) but I ended up not using it since it seemed unstable. Thanks.

loboris, 2018-03-07:
I'll close this issue as the feature is not yet implemented.

closed 2018-03-07 by loboris