loboris/MicroPython_ESP32_psRAM_LoBo #90: Building on Arch requires installing python-pyserial

slzatz, 2018-03-07:
Instructions say install python2-pyserial but my experience is that won't build on Arch without python3 pyserial

loboris, 2018-03-07:
I've newer tested the build on Arch Linux.
The prerequisites for Arch were taken from esp-idf documentation.
If you had to install anything else to make it work on Arch, could you, please, post the needed packages, and I'll update the instructions.

slzatz, 2018-03-07:
Just add `sudo pacman -S python-pyserial` and everything works fine.  The default python on Arch is python3.x.  It's possible that python2 is being called explicitly somewhere so I didn't remove the python2 pyserial although I am not sure it is actually needed.

loboris, 2018-03-07:
Ok, added.

closed 2018-03-07 by slzatz