loboris/MicroPython_ESP32_psRAM_LoBo #91: Useful information  to Wiki related to pye module (Documentation).

djairjr, 2018-03-08:
Boris, 
I've found this useful information about pye module (I'm a python-newbie, and maybe this is kind of obvious, but...):

https://github.com/robert-hh/Micropython-Editor

Can you put this link in Wiki?

djairjr, 2018-03-08:
Source: https://github.com/robert-hh/Micropython-Editor

# Pye - Micropython-Editor
Description:

A small text editor written in Python running on PYBoard, WiPy 1, the pycom.io modules like WipPy 2, Lopy, SiPy, ESP8266 modules and teensy 3.5/3.6, allowing to edit files locally. It is based on the editor widget of pfalcon athttps://github.com/pfalcon/pyedit. I ported it to PyBoard, WiPy 1, ESP8266, Teensy, ESP32 and the Pycom.io devices, and added a few functions:

Use sys.stdin.read() and sys.stdout.write() for input and output of the Micropython version.
Changed the read keyboard function to comply with slow byte-by-byte input on serial lines.
Added support for Tab, BackTab, Save, Del and Backspace joining lines, Find, Replace, Goto Line, Undo, Get file, Auto-Indent, Set Flags, Copy/Delete & Paste, Indent, Un-Indent
Handling tab (0x09) on reading & writing files,
Added a status line, and single line prompts for Quit, Save, Find, Replace, Goto, Get file and Flag settings.
Optional support of the basic mouse functions scrolling up/down, setting the cursor and marking lines.

Warning: Due to a glitch in the pycom.io ports of micropython before V1.9.0 Ctrl-C must not be used for copying in these ports. Use Ctrl-D instead. Unfortunately, Ctrl-C will terminate the editor (Bug catching KeyboardInterrupts Exceptions).

The editor is contained in the file pye.py. Start pye from the REPL prompt e.g. with:

```
from pye import pye
res = pye(object_1, object_2, ..[, tabsize=n][, undo=n])
```

If object_n is a string, it's considered as the name of a file to be edited or a directory to be opened. If it’s a file, the content will be loaded, and the name of the file will be returned when pye is closed. If the file does not exist, an error is displayed, but the edit window is given that name. If it’s a directory, the list of file names will be loaded to the edit window. If object_n is a list of strings, these will be edited, and the edited list will be returned. If no object is named, pye() will give you an empty screen, creating a list of strings, unless you save to a file. In that case, the file name will be returned. If object_n is neither a string nor a list of strings, an empty buffer is opened. It is always the last buffer closed, which determines the return value of pye().

Optional named parameters:
tabsize=n Tab step (integer). The default is 4
undo=n Size of the undo stack (integer). A value of 0 or False disables undo.

The editor assumes a VT100 terminal. It works in Insert mode. The following list shows most of the commands. Commands marked with (opt) may not be supported in minimal versions:

Key(s)                              Function:                                                                                     
Up Down Left Right           Cursor movement by one line or char
PgUp & PgDd                    Page up/down
Home                               Toggle the position between the start-of-code and the start of line
End                                  Toggle the position between the end-of-the-code and end-of-line
Enter                                Enter a line break at the cursor position. Auto-indent is supported
Backspace                         Delete char left to the cursor (The key must be set to ASCII-Del)
Del                                   Delete the char under the cursor. At the end of the line join the next line. 
                                        If autoindent is enabled, delete also the leading spaces of the joined line. 
                                        If lines are marked, delete the marked area
Tab & Backtab                    Insert or remove spaces up to the next tab position. 
                                        If lines are marked, indent or unindent (opt)
Ctrl-O                                Open a new file. If the file name is left empty, an empty buffer is opened
Ctrl-W                               Toggle to the next file buffer
Ctrl-Q or Ctrl-|                   Close a file buffer or end line-edit
Ctrl-S                                Save to file with the option to change the file name
Ctrl-W                               Switch to the next file buffer
Ctrl-F                                Find
Ctrl-N                               Repeat last find
Ctrl-H                               Find and Replace (opt)
Ctrl-G                               Go to a line
Ctrl-T                               Go to the first line (opt)
Ctrl-B                               Go to the last line (opt)
Ctrl-K                               Goto the bracket matching the one under the cursor (opt)
Ctrl-L or Ctrl-Space            Mark/Unmark the current line. The mark can then be extended by moving the cursor
Ctrl-X or Ctrl-Y                  Cut the marked lines
Ctrl-C or Ctrl-D                  Copy the marked lines
Ctrl-V                               Insert the copied/cut lines. 
                                       In the line edit mode paste the item under the cursor of the active window
Ctrl-Z                               Undo the last change(s)
Ctrl-P                               Comment/Uncomment a line or marked area
Ctrl-A                               Change settings for tab size, search case sensitivity, 
                                       auto-indent, comment string and writing tabs (opt)
Ctrl-E                               Redraw the screen. On WiPy and PyBord it shows the amount of free memory

loboris, 2018-03-08:
The pdf manual for pye editor is available in [Documents directory](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/Documents/Pyboard%20Editor.pdf)

loboris, 2018-03-24:
As this is not acctually an issue, and this information is posted on on forum, I'll close it.

closed 2018-03-24 by loboris

bdespatis, 2018-03-28:
Great stuff. It saves so much trouble. I did not have to send my small files by FTP, just telnet and pye editor. Thanks for adding this !