loboris/MicroPython_ESP32_psRAM_LoBo #93: ftp stack overflow while synchronizing

kevinkk525, 2018-03-08:
I'm trying to synchronizie my project files to the esp32 using ftp now, as I at least got it working, (still same setup as #78 ) but during that, the esp crashes after a few files. It does not matter if I upload it with lftp or using filezilla. Last time it uploaded 13 files before it crashed.

```
***ERROR*** A stack overflow in task FtpServer has been detected.
abort() was called at PC 0x40093968 on core 1
0x40093968: vApplicationStackOverflowHook at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

Backtrace: 0x40093788:0x3ffcf2a0 0x4009394f:0x3ffcf2c0 0x40093968:0x3ffcf2e0 0x4008fece:0x3ffcf300 0x40091eec:0x3ffcf320 0x40091ea2:0xb33fffff
0x40093788: invoke_abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4009394f: abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x40093968: vApplicationStackOverflowHook at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4008fece: vTaskSwitchContext at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:4702

0x40091eec: _frxt_dispatch at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./portasm.S:406

0x40091ea2: _frxt_int_exit at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./portasm.S:206

CPU halted.
```

If I try to access the folder where it crashed, it crashes again with the exact same error message.
```
***ERROR*** A stack overflow in task FtpServer has been detected.
abort() was called at PC 0x40093968 on core 1
0x40093968: vApplicationStackOverflowHook at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

Backtrace: 0x40093788:0x3ffcf230 0x4009394f:0x3ffcf250 0x40093968:0x3ffcf270 0x4008fece:0x3ffcf290 0x40091eec:0x3ffcf2b0 0x40091ea2:0x33707365
0x40093788: invoke_abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4009394f: abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x40093968: vApplicationStackOverflowHook at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x4008fece: vTaskSwitchContext at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:4702

0x40091eec: _frxt_dispatch at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./portasm.S:406

0x40091ea2: _frxt_int_exit at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./portasm.S:206

CPU halted.
```
I then have to remove that folder manually using microPython.

Sometimes it crashes in various other subfolders too with the same error. I somehow seem to be the only one having problems with ftp module.
There is nothing else running in the python thread, just loading some configs, wlan connection, rtc sync, ftp start, telnet start, mdns start. just like your startup script, just with loading config first.

loboris, 2018-03-09:
I can reproduce this with more than one subdirectory levels.
I'm working on fixing this issue.
I expect to finish it today or tomorrow.

loboris, 2018-03-15:
This issue should be fixed in the latest update.

kevinkk525, 2018-03-15:
Thanks for putting effort into this.
It's a lot better now but it is not completely reliable. 
Occasionally on folder synchronization and random access it still crashes:

```
/home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:1443 (vTaskDelay)- assert failed!
abort() was called at PC 0x40090f4d on core 1
0x40090f4d: vTaskDelay at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:4702

Backtrace: 0x40093cfc:0x3ffb7a80 0x40093ec3:0x3ffb7aa0 0x40090f4d:0x3ffb7ac0 0x400fe34c:0x3ffb7ae0 0x400fe404:0x3ffb7b00 0x400f03d3:0x3ffb7b20 0x400eb8ad:0x3ffb7b50 0x400eb91d:0x3ffb7b70 0x400f97d9:0x3ffb7b90 0x400f046e:0x3ffb7c30 0x400eb8ad:0x3ffb7c60 0x400eb91d:0x3ffb7c80 0x400f97d9:0x3ffb7ca0 0x400f046e:0x3ffb7d40 0x400eb8ad:0x3ffb7d70 0x400eb91d:0x3ffb7d90 0x400f97d9:0x3ffb7db0 0x400f046e:0x3ffb7e50 0x400eb8ad:0x3ffb7ec0 0x400eb8da:0x3ffb7ee0 0x400f63bc:0x3ffb7f00 0x400f6462:0x3ffb7f90 0x400f677e:0x3ffb7fc0 0x400ec3ad:0x3ffb80a0 0x400f9b36:0x3ffb8100 0x400f046e:0x3ffb81a0 0x400eb8ad:0x3ffb8200 0x400eb8da:0x3ffb8220 0x400de2f5:0x3ffb8240 0x400de5a5:0x3ffb82f0 0x400d7566:0x3ffb8310
0x40093cfc: invoke_abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x40093ec3: abort at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:648

0x40090f4d: vTaskDelay at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:4702

0x400fe34c: poll_poll_internal at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/../extmod/moduselect.c:255

0x400fe404: poll_ipoll at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/../extmod/moduselect.c:293

0x400f03d3: fun_builtin_var_call at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:127

0x400eb8ad: mp_call_function_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400eb91d: mp_call_method_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:643

0x400f97d9: mp_execute_bytecode at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1018

0x400f046e: fun_bc_call at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x400eb8ad: mp_call_function_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400eb91d: mp_call_method_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:643

0x400f97d9: mp_execute_bytecode at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1018

0x400f046e: fun_bc_call at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x400eb8ad: mp_call_function_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400eb91d: mp_call_method_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:643

0x400f97d9: mp_execute_bytecode at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1018

0x400f046e: fun_bc_call at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x400eb8ad: mp_call_function_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400eb8da: mp_call_function_0 at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400f63bc: do_execute_raw_code at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/builtinimport.c:167

0x400f6462: do_load at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/builtinimport.c:207

0x400f677e: mp_builtin___import__ at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/builtinimport.c:461

0x400ec3ad: mp_import_from at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:1378

0x400f9b36: mp_execute_bytecode at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/vm.c:1211

0x400f046e: fun_bc_call at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/objfun.c:267

0x400eb8ad: mp_call_function_n_kw at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400eb8da: mp_call_function_0 at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:613

0x400de2f5: parse_compile_execute at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:531

0x400de5a5: pyexec_file at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/lib/utils/pyexec.c:502

0x400d7566: mp_task at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/esp32/main.c:154

CPU halted.
```

loboris, 2018-03-15:
This is strange.
Are you using the **uftpserver** python module ?
If yes, please use the integrated ftp class from network module ([Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/ftpserver))
That should also solve your issue https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/103

The crash log shows that that the crash originates from _moduselect.c_ which is not used in the integrated ftp server.

kevinkk525, 2018-03-15:
Well that's interesting. I'm not using that module, I'm definitely using the integrated ftp class from network module. It's started up like this, according to your startup instructions:

```
if sys.platform == "esp32_LoBo":
            rtc = machine.RTC()
            print("Synchronize time from NTP server ...")
            rtc.ntp_sync(server="de.pool.ntp.org")  # update once
            tmo = 100
            while not rtc.synced():
                time.sleep_ms(100)
                tmo -= 1
                if tmo == 0:
                    break
            if config.MDNS_ACTIVE:
                mdns = network.mDNS()
                mdns.start(config.MDNS_HOSTNAME, config.MDNS_DESCRIPTION)
            if config.FTP_ACTIVE:
                network.ftp.start()
                if config.MDNS_ACTIVE:
                    mdns.addService('_ftp', '_tcp', 21, "MicroPython", {
                                    "board": "ESP32", "service": "mPy FTP File transfer", "passive": "True"})
            if config.TELNET_ACTIVE:
                network.telnet.start()
                if config.MDNS_ACTIVE:
                    mdns.addService('_telnet', '_tcp', 23, "MicroPython", {
                                    "board": "ESP32", "service": "mPy Telnet REPL"})
```

If the crash log does not have anything to do with the ftp module, then maybe it's a weird coincidence..

kevinkk525, 2018-03-15:
I think I see where the error is. asyncio uses moduselect and ipoll. It's most probably now a problem between ftp task and asyncio task.
Issue #103 still persist though, removing leaves me with a bunch of empty folders

kevinkk525, 2018-03-18:
Closing this as it is a rare problem and I made a thread on the forum about general problems with asyncio and other tasks running: [Weird-errors-with-asyncio](https://loboris.eu/forum/Thread-Weird-errors-with-asyncio)

closed 2018-03-18 by kevinkk525