loboris/MicroPython_ESP32_psRAM_LoBo #97: Feature request: Implement the two missing rtc.wake_on_ext modes.

stroobandt, 2018-03-11:
> **`rtc.wake_on_ext1(pins [,level])`**
> If level is set to 0, all pins must be at low level to wake up.
> If level is set to 1, any pin at high level will wake up.

The absolute icing on the cake would be to have a third mode: **any pin at low level will wake up.**
That would awake a device from `deepsleep` by pressing any button that is pulled up by a resistor.

I am pretty sure someone will also find good use for a remaining fourth mode: **all pins must be at high level to wake up.**

loboris, 2018-03-15:
Wake on **all pins HIGH** is supported in the latest update (using ESP32 wake stub).
ESP32 wakes when any pin is at HIGH level, but the wake stub puts it back to sleep if not ALL pins are HIGH.
Use `level=2` fot that.

Wake on **any pin at low level will wake up** is not possible to implement, sorry.

stroobandt, 2018-03-16:
If I understand correctly; this is a limitation imposed by Espressif on the ESP32?
Perhaps future Espressif updates will make this possible…

loboris, 2018-03-16:
I think the ESP32 hardware does not support it, so it canno't be fixed in esp-idf, but I'm not shure.

loboris, 2018-04-08:
I think all possible wake modes for external pins are now supported, so I'll close the issues.

closed 2018-04-08 by loboris