loboris/MicroPython_ESP32_psRAM_LoBo #98: UART.init (lineend)

bdespatis, 2018-03-12:
Hi,

I tried to do this : 
```
>>> from machine import UART
>>> uart = UART(2,tx=17,rx=16)
D (37623) intr_alloc: Connected src 36 to int 9 (cpu 1)
>>> uart.init(baudrate = 115200, bits =8, parity = None, stop=2, lineend='\r')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given
>>> uart.init(baudrate = 115200, bits =8, parity = None, stop=2) //This works

```
I would like to have \r as lineend. Thanks for the help !

Regards,
Benoit

loboris, 2018-03-16:
The UART module is updated to show the lineend in `print(uart)`, some minor bugs are fixed.
Testing line end:
```python
>>> import machine
>>> uart = machine.UART(2,tx=27,rx=32)
>>> uart
UART(2, baudrate=115201, bits=3, parity=None, stop=1, tx=27, rx=32, rts=-1, cts=-1, timeout=0, buf_size=512, lineend=b'\r\n')
     Event task minimum free stack: 608
>>> uart.init(lineend='\r')
>>> uart
UART(2, baudrate=115201, bits=3, parity=None, stop=1, tx=27, rx=32, rts=-1, cts=-1, timeout=0, buf_size=512, lineend=b'\r')
     Event task minimum free stack: 608
>>> uart.init(lineend='\x12\r')
>>> uart
UART(2, baudrate=115201, bits=3, parity=None, stop=1, tx=27, rx=32, rts=-1, cts=-1, timeout=0, buf_size=512, lineend=b'\x12\r')
     Event task minimum free stack: 608
```

The **buffer size** can be set only when creating the uart instance, not with `uart.init()`.
Default is 512 bytes.
Here is the test with `uart.any()`, more than 512 bytes was sent before `uart.read()`, after the buffer is full, new received bytes are ignored :

```python
>>> uart.any()
0
>>> uart.any()
196
>>> uart.any()
351
>>> uart.any()
476
>>> uart.any()
512
>>> 
>>> uart.any()
512
>>> uart.read()
b'aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa\rbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbbb\rcccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccccc\rdddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddddd\reeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeeee'
˙``

bdespatis, 2018-03-16:
Thanks ! I will test it today.

bdespatis, 2018-03-16:
Strange behavior here with readline. I can init now, but uart.readline() did not returned the proper information.

''''
>>> uart.init(lineend='\r')
>>> uart.any()
9
>>> uart.readline()
b'+CCCCCCCC'
>>> uart.any()
9
>>> uart.readline()
b'+CCCCCCCC'
>>> uart.any()
9
>>> uart.read()
b'+CREG:-2\r'
''''
It's like the function copy the first character ...

loboris, 2018-03-17:
There was a bug in moving the internal uart buffer causing this behaviour, it is fixed now and commited.

BTW, you can also use `uart.readln()` method, there are some differences between it and `uart.readline()`.
`uart.readline()` uses MicroPython streams to read the data. It expects `\n` as the line end and it is also slower than `uart.readln()` which reads the data directly from the uart buffer.
The [Wiki page](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/uart) is updated, you can get more detailed information there.

loboris, 2018-03-24:
I think the issue is fixed.
Please reopen if needed.

closed 2018-03-24 by loboris