loboris/MicroPython_ESP32_psRAM_LoBo #100: CPU halted

GregCle, 2018-03-14:
Hi,

I have an error that rarely happens (about 1 time of 50).

When I start my code, I try to connect to an access point and after I start the mqtt protocol.

During startup, a network is searched and an ip address is found (See picture)

[image]

```
def Init_Connection():
    global Adr_Ip_Asept3X, ID, Ad_Ip_MQTT, Net_Name, Pass
    ID = 'Asept3x_'
    wlan.active(False)
    while wlan.ifconfig()[0] !='0.0.0.0':
        time.sleep(1)

    wlan.active(True)
    wlan.connect(Net_Name,Pass)
    #----- Attend connexion = True ------
    while wlan.isconnected() != True :
        Toggle(LED_Error)
        time.sleep_ms(500)
    time.sleep(2)
    network.telnet.start()
    #-------- Recup茅ration info IP adresse + creation ID
    Adr_Ip_Asept3X = wlan.ifconfig()[0]
    Ad_Ip_MQTT = wlan.ifconfig()[2]
    ID += Adr_Ip_Asept3X.split('.')[3]
    print(('Asept3x IP: ' + Adr_Ip_Asept3X))
    print(('Broker IP: ' + Ad_Ip_MQTT))
    print(('ID: ' + ID))
```

This problem happens a few times and it is hard to recreate.

Do you have an idea how to fix my problem?

Thank you

loboris, 2018-03-18:
It looks like it happens while running the main.py, could you post the complete main.py and more detailes of what the program does?
From the log, it looks like display is running too...

GregCle, 2018-03-19:
Hi, 

I will try to give the maximum detail.
Here are the libraries I use (no display lib):
```
#I modified the libraries to use only the necessary one, but the error is produced using this one
import time
import network
import machine
import os
import micropython
from machine import Pin
from machine import PWM
import _thread
import gc

```
Next:

```
# I check my external eeprom
Init_Memory()

#I get some information from eeprom
Net_Name = Read_Text(16)
Pass = Read_Text(18)

#I check if a particular document is present in the flash, If yes, I read it and get the info
Read_File = os.listdir()

for x in range(0, len(Read_File)):
    if Read_File[x] == 'Parametres.txt':
        file = open('Parametres.txt', 'r')
        ssid = file.readline().rstrip("\r\n")
        mdp = file.readline().rstrip("\r\n")
        Write_Text(16,ssid)
        Write_Text(18,mdp)
        file.close()
        os.remove('Parametres.txt')
        Net_Name = ssid
        Pass = mdp

#If the eeprom is empty, I create an access point
#else I connect to an access point (to use the Mqtt)

if SW.value() == 0 or (Net_Name == '0000000000000000' and Pass == '0000000000000000'):
    AP()
else:
    Nb_Allumage = Read_Number(4)
    Nb_Allumage_Total = Read_Number(5)
    Nb_Erreur_Pir = Read_Number(6)
    Nb_Erreur_Pir_Total = Read_Number(7)
    Nb_Allumage_Systeme = Read_Number(8)

    Nb_Allumage_Systeme = Nb_Allumage_Systeme + 1
    Write_Number(8,Nb_Allumage_Systeme)

    wlan = network.WLAN(0)

    #Connect to the access point
    # It's after the CPU problem appears
    Init_Connection()

    #----- Creation client MQTT
    client = network.mqtt("Broker_IAP", Ad_Ip_MQTT, clientid = ID ,autoreconnect = True,cleansession=True,connected_cb=conncb, disconnected_cb=disconncb, subscribed_cb=subscb, published_cb=pubcb, data_cb=datacb)

    Check_MQTT()

    LED_Error.value(0)

    _thread.start_new_thread('Main',Main, ())
```

Init_Connection() code:

```
def Init_Connection():
    global Adr_Ip, ID, Ad_Ip_MQTT, Net_Name, Pass
    ID = 'Client'
    wlan.active(False)
    while wlan.ifconfig()[0] !='0.0.0.0':
        time.sleep(1)

    wlan.active(True)
    wlan.connect(Net_Name,Pass)
    #----- Attend connexion = True ------
    while wlan.isconnected() != True :
        Toggle(LED_Error)
        time.sleep_ms(500)
    time.sleep(2)
    network.telnet.start()
    #-------- Recup茅ration info IP adresse + creation ID
    Adr_Ip = wlan.ifconfig()[0]
    Ad_Ip_MQTT = wlan.ifconfig()[2]
    ID += Adr_Ip.split('.')[3]
    print(('Client IP: ' + Adr_Ip))
    print(('Broker IP: ' + Ad_Ip_MQTT))
    print(('ID: ' + ID))
```

I can give you the rest of the code, but I did not want to put too much in it and get lost

Thank you

loboris, 2018-03-19:
Your log (on the attached image) shows that the **`dsplay.tft.drawArc`** function from the **display module** was called before crash.

How do you access external eeprom, over I2C? 
BTW, ESP32 has NVS memory which can be used so store the parameters.

You could place some debug prints in various places in your code to locate where the cracs occurs.

It may help if you can send the rest of the code, you can also open the issue thread on the [forum](https://loboris.eu/forum/)

GregCle, 2018-03-20:
Hi Loboris,

Here is the complete code.

I use an external eeprem over I2C. We have added an external eeprom for the very large write count (time registration + recovery code if electrical failure during writing)

Yesterday too I had a halted CPU. This time the problem happened just after sending a command via the MQTT.

I'm sorry to put photos, but the ESP32 is connected to a Raspberry Pi that I do not have easy access to copy the text.

[image]

```
import time
import network
from machine import reset
import os
from machine import Pin
from machine import PWM
import _thread
import gc

gc.threshold(2000)

LED_Error = Pin(13,Pin.OUT)

SW = Pin(33,Pin.IN, Pin.PULL_UP)

Pir = Pin(15,Pin.IN,Pin.PULL_DOWN)

Triac_1 = Pin(32,Pin.OUT)
Triac_2 = Pin(23,Pin.OUT)
Triac = [Triac_1,Triac_2]

Fan_1 = PWM(Pin(22))
Fan_2 = PWM(Pin(19))
Fan_1.freq(500)
Fan_2.freq(500)
Fan_1.duty(0)
Fan_2.duty(0)
Fan = [Fan_1, Fan_2]

L1 = Pin(14,Pin.IN)
L2 = Pin(12,Pin.IN)
L3 = Pin(5,Pin.IN)
L4 = Pin(4,Pin.IN)

STATE = 'STAND_BY'
STAND_BY = 'STAND_BY'
TURN_ON_FAN = 'TURN_ON_FAN'
TURN_OFF_FAN = 'TURN_OFF_FAN'
ON_LAMP = 'LAMP_ON'
PIR_DETECTION = 'PIR_DETECTION'
OFF_LAMP = 'LAMP_OFF'
RESET = 'RESET'

ACTION = ['Lamp_On', 'Lamp_Off', 'Fan_On', 'Fan_Off', 'Stand_By', 'No_Safe', 'State', 'Reset']

Lamp = [L1,L2,L3,L4]
Lamp_Work = ['1','1','1','1']
Lamp_Doesnot_Work = ['0','0','0','0']
Lamp_Stop = ['0','0','0','0']
Status_Lamp = [' ',' ',' ',' ']

#------ MQTT -----
ID = 'Client_'
Ad_Ip_MQTT = '192.168.13.63'
Sub_Folder = 'Class1/Lab/Action'
Pub_Folder = 'Class1/Lab/Client_1/State'
Pub_Ip = ''
[... 299 lines cut ...]

if SW.value() == 0 or (Net_Name == '0000000000000000' and Pass == '0000000000000000'):
    AP()
else:
    Nb_Allumage = Read_Number(4)
    Nb_Allumage_Total = Read_Number(5)
    Nb_Erreur_Pir = Read_Number(6)
    Nb_Erreur_Pir_Total = Read_Number(7)
    Nb_Allumage_Systeme = Read_Number(8)

    Nb_Allumage_Systeme = Nb_Allumage_Systeme + 1
    Write_Number(8,Nb_Allumage_Systeme)

    wlan = network.WLAN(0)
    Init_Connection()

    #----- Creation client MQTT
    client = network.mqtt("Broker", Ad_Ip_MQTT, clientid = ID ,autoreconnect = True,cleansession=True,connected_cb=conncb, disconnected_cb=disconncb, subscribed_cb=subscb, published_cb=pubcb, data_cb=datacb)

    Check_MQTT()

    LED_Error.value(0)

    _thread.start_new_thread('Main',Main, ())
```

loboris, 2018-04-08:
Do you still have this issue after the latest commits ?

closed 2018-05-17 by loboris