loboris/MicroPython_ESP32_psRAM_LoBo #101: ftp server not working when in AP mode

pidou46, 2018-03-15:
Hello Boris,

I think it would be very convenient to be able to run the ftp and telnet server in AP mode, no need to be connected to local WIFI.
But FTP server refuse to start when network is set to AP, I get the 'Invalid Wifi mode' message
I think it's more or less related to "MicroWebSrv not started when in AP mode #67"

>>> import network
>>> import time
>>> ap_if=network.WLAN(network.AP_IF)
>>> ap_if.active(True)
>>> time.sleep(6)
>>> network.ftp.start(user="micro", password="python", buffsize=1024,timeout=300)
E (2525205) [Ftp]: Invalif WiFi mode

Do you think you could allow it ?

Thanks

loboris, 2018-03-18:
Thanks for the suggestion, I will test it and enable if it works good.

pidou46, 2018-03-18:
Nice, thank you to have a look.

It would help the workflow.

loboris, 2018-04-08:
As this is working (for some time), I'll close the issue.

closed 2018-04-08 by loboris