loboris/MicroPython_ESP32_psRAM_LoBo #103: ftp not removing folders

kevinkk525, 2018-03-15:
Now that I can synchronize folders I recognized, that I'm unable to delete any folder.

Is this a bug or just not completely implemented?

loboris, 2018-03-16:
Ive tested with FileZilla and it removes the folders.

It is up to the ftp client to first remove all the files in the directory, and thed remove the (now empty) directory itself.

kevinkk525, 2018-03-16:
ok, I didn't know that the client is responsible for that. I also use filezilla as described in your wiki but the folders are not removed. 
Is it something with the filesystem on the µC?

kevinkk525, 2018-03-16:
Can confirm that it is fixed. Only weird things I still got is that some folders can't get deleted, not even by os.rmdir..

closed 2018-03-16 by kevinkk525