loboris/MicroPython_ESP32_psRAM_LoBo #104: New Commit Build error

elektrobil, 2018-03-16:
Hi @loboris 

I pulled the last commit but failed. Build error message belaow.

> home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c: Assembler messages:
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:858: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:861: Error: unknown opcode or format name 'wsr.INTENABLE'
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:876: Error: unknown opcode or format name 'xsr.INTENABLE'
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:879: Error: unknown opcode or format name 'wsr.INTENABLE'
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/component_wrapper.mk:273: recipe for target 'intr_alloc.o' failed
make[1]: *** [intr_alloc.o] Error 1
/home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project.mk:451: recipe for target 'component-esp32-build' failed
make: *** [component-esp32-build] Error 2
'make all' FAILED!

kevinkk525, 2018-03-16:
Did you change something in menuconfig?
Did you make clean before building?
I always get weird problems either while building or in micropython if I don't make clean after pulling updates.

elektrobil, 2018-03-16:
I pulled and enter menu config make my standart setup its failed and change directory name and cloned and again build failed error appears same lines

loboris, 2018-03-16:
@elektrobil

I've just tried to update the older version (`git pull`) and I was able to build without problem.

After successfull `git pull` anways run `./BUILD.sh menuconfig`.

If the _esp-idf_ or _toolchain_ was changed (like in the last commit), you will get the message:
```
Removing old tools versions and cleaning build...
unpacking 'esp-idf'
unpacking Linux/xtensa-esp32-elf
```
before menuconfig is run, and the clean will be run automatically.
If not, always run `./BUILD.sh clean` before build.

If you can't get it working, you can always execute the full clone into the new directory.
`git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git`

elektrobil, 2018-03-16:
I download the full clone into a new directory and first-- run menuconfig make setups and after exit then clean command and run ./Build.sh all -v and same error occurs

loboris, 2018-03-16:
On which OS are you building ?

Here is the full log of fresh clone and build on Ubuntu 17.10:

```
boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test$ git clone https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo.git
Kloniranje u 'MicroPython_ESP32_psRAM_LoBo'...
remote: Counting objects: 3033, done.
remote: Compressing objects: 100% (32/32), done.
remote: Total 3033 (delta 9), reused 23 (delta 7), pack-reused 2988
Primanje objekata: 100% (3033/3033), 636.91 MiB | 1.97 MiB/s, završeno.
Razrješavanja delta: 100% (1349/1349), završeno.
boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test$ 
boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test$ 
boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test$ cd MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD
boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ ./BUILD.sh menuconfig
Removing old tools versions and cleaning build...
unpacking 'esp-idf'
unpacking Linux/xtensa-esp32-elf
make[1]: Entering directory '/home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD -lncursesw -ltinfo
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/util.o lxdialog/util.c
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncursesw -ltinfo
cc  -D_GNU_SOURCE -I/usr/include/ncursesw -DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1 -DLOCALE -MD   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncursesw -ltinfo
make[1]: Leaving directory '/home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/tools/kconfig'
DEFCONFIG
#
# configuration written to /home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/sdkconfig
#
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

GENCONFIG

---------------------
MicroPython for ESP32
---------------------

MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

OK.

boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ ./BUILD.sh -j8 all

---------------------
MicroPython for ESP32
---------------------

==================
Building mpy-cross
OK.
==================
================================
Building MicroPython firmware...
================================
Restarting build ...
OK.
--------------------------------
Build complete.
You can now run ./BUILD.sh flash
to deploy the firmware to ESP32
--------------------------------

boris@UbuntuMate:/home/LoBo2_Razno/ESP32/MicroPython/novi_test/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD$ 
```

elektrobil, 2018-03-16:
I'm using Opensuse Leap 42.3

loboris, 2018-03-16:
There are no specific prerequisites for OpenSuse listed by Espressif, but it should work.
Please check if you hawe all the packages installed, listed for other distributuins ([Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/build)).

If you were able to build the firmware before the latest commit (have you?), then probably something changed in the **esp-idf** which is causing the error.

elektrobil, 2018-03-16:
libncurses-devel package not found appear when run Install Prerequisites command. Opensuse uses name"ncurseslib" not 'libncurses' is that any changes for build ?

loboris, 2018-03-16:
Try to install **ncurses** and **ncurses-devel** using your package manager.
Also make shure all other packages (or their OpenSuse equivalents) are installed.

elektrobil, 2018-03-16:
all of installed. Previous version works good and build okay but this commit has that error
`
home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c: Assembler messages: /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:858: Error: unknown opcode or format name 'xsr.INTENABLE' /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:861: Error: unknown opcode or format name 'wsr.INTENABLE' /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:876: Error: unknown opcode or format name 'xsr.INTENABLE' /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./intr_alloc.c:879: Error: unknown opcode or format name 'wsr.INTENABLE' /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/component_wrapper.mk:273: recipe for target 'intr_alloc.o' failed make[1]: *** [intr_alloc.o] Error 1 /home/cem/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/make/project.mk:451: recipe for target 'component-esp32-build' failed make: *** [component-esp32-build] Error 2 'make all' FAILED!`

loboris, 2018-03-16:
I'll try to install Opensuse Leap 42.3 in VM to check this issue.

loboris, 2018-03-16:
I've installed Opensuse Leap 42.3 (Gnome) in VM.

Insatlled prerequisites with
```
sudo zypper install gcc git wget make ncurses-devel flex bison gperf python python-serial
```
Cloned and built the same way as on Ubuntu (see one of my previous posts).

No error was reported.

elektrobil, 2018-03-16:
Thank you so much I'll try and I'm going to write result

elektrobil, 2018-03-17:
Hi @loboris 
I found the problem. I changed OS languange to Turkish its because Toolchain give error I fix it. Its Working Thank You So Much

closed 2018-03-17 by elektrobil

loboris, 2018-03-17:
I'm glad you solved the issue.
It is still strange, I have the OS language set to Croatian and have no errors ...