loboris/MicroPython_ESP32_psRAM_LoBo #106: Neopixel.timings method bug

rdagger, 2018-03-17:
The Neopixel.timings() method argument order is wrong but more seriously it generates pulses that are out of order.  The faulty pattern looks like: 
T1L always precedes T0H  instead of T1L following T1H.
and T0L always precedes T1H instead of T0L following T0H. 
Basically the lows are out of order.
[image]

loboris, 2018-03-18:
Thanks, I'll review the Neopixel timings next week.

wangshujun-tj, 2019-02-14:
Neopixel and apa102 recommend using SPI module driver, setting the bps to 8 megabits and transferring 1 bit（ws2812） per byte（SPI）, which is very stable.
In addition, the timing of WS2812 is very insensitive to low-level stretching, just do not exceed reset time.