loboris/MicroPython_ESP32_psRAM_LoBo #107: Crash with unknown reason

kevinkk525, 2018-03-19:
I keep getting a crash but the reason is completely unknown to me as the error message does not contain any information that is useful to me.
I'm not doing anything special where it occured, just running my normal mqtt asyncio code with RTC synchronization enabled, mdns & ftp disabled.
sdkconfig is attached.
[sdkconfig.txt](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1825309/sdkconfig.txt)

```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr
Core 0 register dump:
PC      : 0x400803c0  PS      : 0x00060336  A0      : 0x00000001  A1      : 0x3ffaf8c0
0x400803c0: _DoubleExceptionVector at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./xtensa_vectors.S:512

A2      : 0x00000003  A3      : 0x00060023  A4      : 0x00060023  A5      : 0x3ffaf970
A6      : 0x00000000  A7      : 0x3ffafa98  A8      : 0x00000001  A9      : 0x00000001
A10     : 0x00060120  A11     : 0x00000000  A12     : 0x800905e8  A13     : 0x3ffaf960
A14     : 0x00000003  A15     : 0x00060123  SAR     : 0x0000001a  EXCCAUSE: 0x00000001
EXCVADDR: 0xffffffd1  LBEG    : 0x40099460  LEND    : 0x4009946b  LCOUNT  : 0xffffffff
0x40099460: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4009946b: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x400803c0:0x3ffaf8c0 0x7ffffffe:0x3ffb7aa0 0x400eb8f6:0x00000017
0x400803c0: _DoubleExceptionVector at /home/kevin/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./xtensa_vectors.S:512

0x400eb8f6: mp_load_method_maybe at /home/kevin/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/runtime.c:1063

CPU halted.
```

loboris, 2018-03-19:
How does your normal "normal mqtt asyncio code" look like.
If you are using the mqtt module, why you have to use acyncio?
The mqtt client already runs in its own FreeRTOS task and has callbacks to inform you of various events.

Asincio is great for some other MicroPython platforms, but I've decided not to support it on this MicroPython port, as I cannot find any reason to use it. Asynchronusly running Python tasks can be defined in a much simpler way using threads and callbacks.

kevinkk525, 2018-03-19:
I could post the whole code but it's basically just an asynchronous mqtt library (not the one ported from c) that keeps me connected and checks for new message. then there is only a regular publish of free ram. that's all that is running. could post the code but it's not small and divided in many packages..

if you don't have an idea by that error message then it's probably not worth trying anyways, as the problem could be anywhere.

Sorry to bother you with asyncio all the time, your port is the only one supporting psram so I do not really have another possibility. 
I can't rewrite my project without asyncio as I use it for the esp8266 as well which does not support threads.

loboris, 2018-03-19:
I'm sorry I could not help.
I would like to make the asyncio code run on this port, but at the moment I have very limited free time.

You could help me if you can provide couple of full asyncio examples (running and not running on this port).
What asyncio version are you using (I think there is more than one) ?

kevinkk525, 2018-03-19:
That's ok, it was not an obvious error message anyways.
That's absolutely understandable, we all develop in our free time and must choose priorities.

I'll see what I can find out. But generally it's not easily reproducible and occurs randomly so I guess it's some problem within the usage of threads, similar to the weird asyncio errors I posted on the forum when using ftp additionally.

As a side question: is it possible to run all threads on one core and only micropython main thread on the other core? As close as possible to the original implementation of micropython? (I have no idea how that was implemented) I guess asyncio should be a lot more stable in that kind of environment having absolute control over the whole core.

eflukx, 2018-03-19:
AFAIK Micropython is ran as a normal RTOS task. Tasks can be pinned to a core using `  xTaskCreateStaticPinnedToCore()`

in original micropython source, see: https://github.com/micropython/micropython/blob/master/ports/esp32/main.c#L121

kevinkk525, 2018-03-20:
I'd like to give a little update as I thought a lot about this problem:
I changed the building configs to "run FreeRTOS only on first core" to active which disables "Use both cores for MicroPython tasks" as well.
Since that change I did not have a single problem with intensively using ftp server while my asyncio program is running. 

I will test this more intensively of course but I thought it would be worth sharing that change as it narrows the problem down significantly. We can now assume that it has something to do with running freeRTOS on both cores (however that is implemented).

loboris, 2018-03-20:
@kevinkk525 
I've just realized that you are running with "Use both cores for MicroPython tasks" enabled, which could be the cause of the problem.

Could you try to run with "run FreeRTOS only on first core" **disabled** (default, means that FreeRTOS runs on both cores) and "Use both cores for MicroPython tasks" **disabled** (which is the **default** setting, and means that the MicroPython task runs on a single core).

With this (default) configuration, MicroPython FreeRTOS task is created pinned to core 1 (App), while various ESP32 system tasks (like WiFi) can run on core 0 (Pro), which increases the performance.

kevinkk525, 2018-03-20:
Seems a lot more stable now. Sometimes the ftp server hangs but mostly catches itself after some time. 
Will close this issue if nothing comes up during the next days and will leave a note on the forum about not using both cores for micropython.

eflukx, 2018-03-20:
Switching to single-core mode seems like a workaround for now, but wouldn't it be feasible to find and fix the root cause?

Is the issue caused solely in "python land" or through a fault in the interaction between micropython and things running (natively) in other threads/cores (e.g. a callback from mqtt client).

Someone able to shed some light on this?

kevinkk525, 2018-03-20:
Only micropython is running single-core. freeRTOS is running on both cores so the wifi etc is running on the core where micropython is not running. 
I do not know how running micropython on both cores was implemented anyways. 

The issue must have been somewhere in threads/cores but not in mqtt client as I did not use the mqtt client by loboris. I use a pure python implementation that is async.

eflukx, 2018-03-20:
Clear, I used the mqtt callbacks (lobo style mqtt that is) as an example only (of interaction accross threads).

Would this issue exist only on the LoBo fork of Micropython (as it includes services running on threads outside the Micropython thread) or would the "mainline" Micropython be equally affected (although I doubt the mainline build of Micropython is configured to run on both cores).

@kevinkk525 I'm planning to build an async application myself on ESP32 Micropython, so anything regarding this has my interest. In that regard Peter Hinch's mqtt async client looks promising. Is that what you're using as mqtt client?

kevinkk525, 2018-03-20:
I did not test the Esp32 mainline port but I also doubt that there is a possibility to run micropython ok both cores. 

Good luck with your async application. I changed his mqtt client a bit to fit the loboris port: https://github.com/kevinkk525/micropython-mqtt/tree/module?files=1
The master branch is the same, the changes are in the module branch.

closed 2018-03-23 by kevinkk525