loboris/MicroPython_ESP32_psRAM_LoBo #109: Asyncio generator error

Lisa999, 2018-03-24:
> Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "upyeasy/__init__.py", line 56, in main
  File "picoweb/__init__.py", line 275, in run
  File "uasyncio/core.py", line 138, in run_forever
  File "uasyncio/core.py", line 107, in run_forever
AttributeError: 'generator' object has no attribute 'pend_throw'

Since micropython is running on a single (App) core, Asyncio IS important!

kevinkk525, 2018-03-24:
Maybe you can give us some more information than an Exception and a statement about asyncio?

Like what version of loboris do you use? What version of asyncio?

can you provide a small code? What happens if you run:
```
import uasyncio as asyncio
loop=asyncio.get_event_loop()
async def test():
   print("test")
loop.run_until_complete(test())
```

Noboy will walk through all the modules written in the Exception etc...

Lisa999, 2018-03-26:
This worked:
`import picoweb, uasyncio as asyncio

app = picoweb.WebApp(__name__)

@app.route("/")
def index(req, resp):
    yield from picoweb.start_response(resp)
    yield from resp.awrite("This is webapp #1")

class plugins(object):
    def __init__(self) :
        print("init")
        
    async def asyncdevices(self):
        print("async")

loop = asyncio.get_event_loop()
_plugins = plugins()
loop.create_task(_plugins.asyncdevices())
app.run(host="0.0.0.0", port=80,debug=False)
`
So uasyncio is working on main level. After moving my loop.create_task from another file into the main subroutine, it worked! Don't know why, but hey i go for the work-around.

Lisa999, 2018-03-26:
I'll test this code:

> import picoweb, uasyncio as asyncio

app = picoweb.WebApp(__name__)

@app.route("/")
def index(req, resp):
    yield from picoweb.start_response(resp)
    yield from resp.awrite("This is webapp #1")

class plugins(object):
    def __init__(self) :
        print("init")
        
    async def asyncdevices(self):
        print("async")

loop = asyncio.get_event_loop()
_plugins = plugins()
loop.create_task(_plugins.asyncdevices())
app.run(host="0.0.0.0", port=80,debug=False)

Lisa999, 2018-03-27:
Same result:

> >>> import upyeasy_async_test
> init
> async
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "upyeasy_async_test.py", line 20, in <module>
>   File "picoweb/__init__.py", line 275, in run
>   File "uasyncio/core.py", line 138, in run_forever
>   File "uasyncio/core.py", line 107, in run_forever
> AttributeError: 'generator' object has no attribute 'pend_throw'
> >>> 
>

kevinkk525, 2018-03-27:
Is upyeasy_async_test the code I posted? You still have not answered what version of uasyncio and uasyncio-core you use, also what version of loboris fork you are on.
and please attach the sdkconfig as sdkconfig.txt

Lisa999, 2018-03-27:
upyeasy_async_test does contain the code above.
Your test code works fine.
Standard loboris firmware:
```
>>> 
ESP32: soft reboot

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1048576; Flash address: 0x200000]
----------------
Filesystem size: 956416 B
           Used: 686080 B
           Free: 270336 B
----------------

FreeRTOS running on BOTH CORES, MicroPython task started on App Core.
Running from Factory partition starting at 0x10000, [MicroPython].

 Reset reason: RTC WDT reset
    uPY stack: 19456 bytes
     uPY heap: 3073664/6160/3067504 bytes (in SPIRAM using heap_caps_malloc)

MicroPython ESP32_LoBo_v3.1.27 - 2017-03-21 on ESP32 board with ESP32
Type "help()" for more information.
>>> 
```

Uasyncio version:
```
srctype = micropython-lib
type = package
version = 1.4.2
author = Paul Sokolovsky
desc = Lightweight asyncio-like library for MicroPython, built around native Python coroutines.
long_desc = README.rst
depends = uasyncio.core
```

I'll test using one of the uasyncio webserver examples

Lisa999, 2018-03-27:
**file upyeasy_async_test3:**
```
import uasyncio as asyncio

@asyncio.coroutine
def serve(reader, writer):
    print(reader, writer)
    print("================")
    print((yield from reader.read()))
    yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\nHello.\r\n")
    print("After response write")
    yield from writer.aclose()
    print("Finished processing request")

class plugins(object):
    def __init__(self) :
        print("init")
        
    async def asyncdevices(self):
        print("async")

import logging
logging.basicConfig(level=logging.DEBUG)
loop = asyncio.get_event_loop()
_plugins = plugins()
loop.create_task(_plugins.asyncdevices())
loop.call_soon(asyncio.start_server(serve, "0.0.0.0", 8081))
loop.run_forever()
loop.close()
```

**Result:**
```
import uasyncio as asyncio

@asyncio.coroutine
def serve(reader, writer):
    print(reader, writer)
    print("================")
    print((yield from reader.read()))
    yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\nHello.\r\n")
    print("After response write")
    yield from writer.aclose()
    print("Finished processing request")

class plugins(object):
    def __init__(self) :
        print("init")
        
    async def asyncdevices(self):
        print("async")

import logging
logging.basicConfig(level=logging.DEBUG)
loop = asyncio.get_event_loop()
_plugins = plugins()
loop.create_task(_plugins.asyncdevices())
loop.call_soon(asyncio.start_server(serve, "0.0.0.0", 8081))
loop.run_forever()
loop.close()
```

It looks like there is a bug...

kevinkk525, 2018-03-27:
do you have an esp8266 to test if it is really a bug in loboris fork? (which i doubt by that exception)

Lisa999, 2018-03-27:
File from micropython-lib: uasyncio/**example_http_server.py:**
```
import uasyncio as asyncio

@asyncio.coroutine
def serve(reader, writer):
    print(reader, writer)
    print("================")
    print((yield from reader.read()))
    yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\nHello.\r\n")
    print("After response write")
    yield from writer.aclose()
    print("Finished processing request")

import logging
#logging.basicConfig(level=logging.INFO)
logging.basicConfig(level=logging.DEBUG)
loop = asyncio.get_event_loop()
loop.call_soon(asyncio.start_server(serve, "127.0.0.1", 8081))
loop.run_forever()
loop.close()
```
**Result:**
>>> import example_http_server
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "example_http_server.py", line 19, in <module>
  File "uasyncio/core.py", line 138, in run_forever
  File "uasyncio/core.py", line 107, in run_forever
AttributeError: 'generator' object has no attribute 'pend_throw'
>>>
```

Standard uasyncio example doesn't work either...

Lisa999, 2018-03-27:
File from micropython-lib: uasyncio/**example_http_client.py:**
```
>>> import example_http_client
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "example_http_client.py", line 23, in <module>
  File "uasyncio/core.py", line 149, in run_until_complete
  File "uasyncio/core.py", line 138, in run_forever
  File "uasyncio/core.py", line 111, in run_forever
AttributeError: 'generator' object has no attribute 'pend_throw'
>>> 
```

kevinkk525, 2018-03-27:
It could be possible that all these things are based on uasyncio-core v2.0 which is not yet compatible with this fork. To test that you would need to run this on an esp8266 with uasyncio-core v2.0 or ask pfalcon about it.

Lisa999, 2018-03-27:
Loboris ESP32 micropython is **NOT** compatible with micropython-lib/uasyncio....
Since i want my program to run on ANY micropython platform, or at least on all ESP32's!, Loboris ESP32 micropython is not usable in the current state. You can hardly expect me to make micropython versions for each platform, other then same very small changes due to the hardware differences, what's the use of micropython then?

Lisa999, 2018-03-27:
Used uasyncio core metadata.txt:
```
srctype = micropython-lib
type = package
version = 1.7.2
author = Paul Sokolovsky
desc = Lightweight asyncio-like library for MicroPython, built around native Python coroutines. (Core event loop).
long_desc = Lightweight asyncio-like library for MicroPython, built around native Python coroutines. (Core event loop).
```
1.7.2 , so it doesn't look like core2...

kevinkk525, 2018-03-27:
It IS compatible with uasyncio, however only versions before v2.0.
There is a development gap between these forks as loboris has a different priority list and added a massive amount of features. He plans to support this but nobody can say when. 

But if it is really an issue that you need V2.0 then just use the official port of esp32...

Lisa999, 2018-03-27:
You're refering to pfalcon/micropython-lib, which i don't use. The pfalcon uasyncio has this metadata.txt:
```
srctype = micropython-lib
type = package
version = 2.0
author = Paul Sokolovsky
desc = Lightweight asyncio-like library for MicroPython, built around native Python coroutines. (Core event loop).
long_desc = Lightweight asyncio-like library for MicroPython, built around native Python coroutines. (Core event loop).
```
As you can see that's the unofficial pfalcon version2 fork, which i'm NOT using...

kevinkk525, 2018-03-27:
if I have time tomorrow I will try your code on my esp32 and esp866.

Lisa999, 2018-03-27:
Don't get me wrong, i would LOVE to use Loboris ESP32 micropython. But it must support the standard micropython language, or else i will get nuts maintaining a lot of different uPyEasy versions [](([https://www.letscontrolit.com/forum/viewtopic.php?f=20&t=3577](url))

kevinkk525, 2018-03-27:
I won't hold you here...
BTW: I am in the process of publishing a big project that is supposed to run on esp8266 and esp32 using uasyncio and i have no problem with loboris fork (still have to do a lot of testing for stability etc but it works). I don't need version 2.0 of uasyncio-core.

Lisa999, 2018-03-27:
I don't need version 2.0 of uasyncio-core either, nor am i using it. The problems with Loboris supporting uasyncio seems to be in the webserver/client area, maybe this part is never tested...

kevinkk525, 2018-04-03:
With the latest commit this fork is now up to date with the main micropython-core. so maybe you could reconsider it now. (sorry did not have time to test your code though)

Lisa999, 2018-04-03:
I'll give it a try and put the results here.

Lisa999, 2018-04-04:
Results:
```
>>> import example_http_server
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "example_http_server.py", line 19, in <module>
  File "uasyncio/core.py", line 146, in run_forever
  File "uasyncio/core.py", line 101, in run_forever
  File "uasyncio/__init__.py", line 235, in start_server
TypeError: function takes 2 positional arguments but 4 were given
>>> 
```
This line in asyncio is the problem:
`ai = _socket.getaddrinfo(host, port, 0, _socket.SOCK_STREAM)`

Easily repeated:

> >>> import usocket as _socket
> >>> ai = _socket.getaddrinfo("www.micropython.org", 80, 0, _socket.SOCK_STREAM)
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
> TypeError: function takes 2 positional arguments but 4 were given
>

Lisa999, 2018-04-04:
I'll open a new call...

mkarliner, 2018-04-04:
I've been through this joy a while back. 
This seems to be caused by a ridiculous state of governance of micropython in general.
It's made me give up on uPy for considerable periods.
May I suggest that the Lobo fork has a policy of being based of the official repos where possible and to clearly state where it doesn't?