loboris/MicroPython_ESP32_psRAM_LoBo #110: Binary blobs in main repo

eflukx, 2018-03-26:
The repo of this port of Micropython carries a lot of binary files: the release firmware files, 6 versions of the xtensa toolchain and a (probably patched) version of the esp-idf.

Although it might make sense to have everything "neatly in one package", the initial `git clone` is currently weighing in at just below 1GiB (!). Especially if a lot of the (firmware) binary files are changed often, this is getting out of hand.

Proposal is to move a away the binary files to separate repositories and having only the project _code_ in this repo.

To have an _EZ2Use_™️ build system maybe have `BUILD.sh` cURL the needed files from the net instead of decompressing a repo-local tar file. Also for pre-build firmware files, why not use the github releases functionality?

kevinkk525, 2018-03-26:
I would appreciate moving the firmware binaries to the release functionality as this would also give the opportunity to have a different development state in the repo and the releases, making it possible to push updates and merge pull requests and build a new firmware from time to time. Also it does not make sense for me to pull all the firmware binaries after every commit as I am building my own firmware from the repo.

As for the esp-idf files I don't see a problem there. The git clone can download that 1GB, it's not the first repo that takes a lot of time cloning. And it is not updated often.
If you get these by curl you would just move that time to the first building command. That's not an advantage in my opinion.

eflukx, 2018-03-26:
Sure more git repo's weigh in at above 1GB, but that doesn't mean I _applaud_ this 'feature' 😜 Might be a matter of personal preference though.

Having (patched) esp-idf in different repo makes sense for instance when having multiple forks of this repo on your PC.

Furthermore when MP-LoBo gets back on mainline ESP-IDF (i.e. unpatched), you don't have to eternally carry the weight of ESP-IDF in your .git history.

kevinkk525, 2018-03-26:
That's true. An additional repo with the esp-idf could make sense.

Let's see what loboris thinks.

loboris, 2018-03-27:
Thank you for the proposals.

I will definitely move the firmware binaries to the separate repository (or to my server), I was planning to to it for a long time, but never found the time to do it.

I was considering  having the **esp-idf** in a separate repository (and make it the git submodule), but for now it is more complicated for me to maintain the repository than to apply couple of patches to esp-idf when it is updated. It will probably be implemented that way in the future.

The **toolchain** binaries are changed infrequently, so I don't think it is an issue.

loboris, 2018-04-08:
I'll do some repository reorganization when the time allows.

For now I'll close the issue, please reopen if needed.

closed 2018-04-08 by loboris