loboris/MicroPython_ESP32_psRAM_LoBo #111: Setting pyhton path does not (fully) work

eflukx, 2018-03-26:
After issuing `./BUILD.sh` on my Arch machine I got this:
```
----------------------------------------
MicroPython for ESP32 with psRAM support
----------------------------------------

==================
Building mpy-cross
FAILED
==================
```

Not a very helpful message indeed. But anyway, after some digging I found that mpy-cross wouldn't build because the `python` executable couldn't be found. That is correct, as my python executable is actually called `python2` (as is the default in Arch).

In `menuconfig` there's an option to set the Python 2 interpreter, but seemingly this config option isn't followed further down the build process.

(quick fix was: `sudo ln -s /usr/bin/python2 /usr/bin/python`)

slzatz, 2018-03-26:
Default python in arch is python3 and /usr/bin/python points to python3. 
 See [arch wiki](https://wiki.archlinux.org/index.php/python)

eflukx, 2018-03-26:
@slzatz That's correct indeed. In my case this was no problem, as I don't have Python 3 installed on this machine.

loboris, 2018-03-27:
To get the verbose output use the **-v** option, as mentioned in the [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/build).
`./BUILD.sh -v all`

loboris, 2018-04-08:
I'll close the issue as there was no activity for some time.
Please reopen if needed

closed 2018-04-08 by loboris