loboris/MicroPython_ESP32_psRAM_LoBo #112: Stack Overflow in task Main

bdespatis, 2018-03-27:
Anyone can tell what is going on ?
MQTT Subscribe was the next step and publish (it was our 66 reboot in a row without failure)

````
assertion "ATB_GET_KIND(bl) == AT_FREE" failed: file "/[...]/[...]/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropyth***ERROR*** A stack overflow in task Main has been detected.
abort() was called at PC 0x4008ead0 on core 1
0x4008ead0: bt_get_i2c_data at ??:?

Backtrace: 0x4008e8f0:0x3ffd28d0 0x4008eab7:0x3ffd28f0 0x4008ead0:0x3ffd2910 0x4008b2f6:0x3ffd2930 0x4008d0a4:0x3ffd2950 0x4008d05a:0x00000000
0x4008e8f0: bt_get_i2c_data at ??:?

0x4008eab7: bt_get_i2c_data at ??:?

0x4008ead0: bt_get_i2c_data at ??:?

0x4008b2f6: xTaskNotify at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./tasks.c:4835

0x4008d0a4: vApplicationStackOverflowHook at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:572

0x4008d05a: esp_panic_wdt_start at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./panic.c:572

CPU halted.
````

No Bluetooth module are added to this config.

Thanks !

bdespatis, 2018-03-27:
````
FreeRTOS running on BOTH CORES, MicroPython task running on both cores.
Running from OTA_0 partition starting at 0x10000, [MicroPython_1].

 Reset reason: Power on reset
    uPY stack: 64512 bytes
     uPY heap: 3501888/2032/3499856 bytes (in SPIRAM using heap_caps_malloc)
````

I found in the menuconfig file that memory was not allocated on both core. Could it be the reason why ? I will test again now.

kevinkk525, 2018-03-27:
Try that and also disable micropython running on both cores. that is not really supported now and a source for possible errors. (in asyncio it does do a lot of errors)

bdespatis, 2018-03-27:
Even if I uncheck Use both cores for MicroPython task, I have the same 

````
FreeRTOS running on BOTH CORES, MicroPython task running on both cores.
````

Very strange ... Can you check on your side if it does the same thing
FreeRTOS : Both Cores
MicroPython : Uncheck both Cores
Allocate memory on both core

Thanks !

kevinkk525, 2018-03-27:
it works for me. that's how it's configured.

try running ./BUILD.sh clean  before building after every change in menuconfig

bdespatis, 2018-03-27:
That's what I am doing all the time ...

I even tried to manually add =n and still, it does not work. Very strange.

kevinkk525, 2018-03-27:
Try my sdkconfig but change the serial port for flashing as I use a different one (running it on WSL).
Remove the file extension .txt 
had to add it because github does not let me upload a file without extension..
[sdkconfig.txt](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1852797/sdkconfig.txt)

bdespatis, 2018-03-27:
I have a new clone of this repository and it works.

Also I found my Saint Graal !!! I found how to reboot in case of panic handler. It was a setting in ESP32-specific. It is not solving all, but a great fix for production ready firmware.

I will test with this new sdkconfig and see how it goes. I can see if it reboot, but at least it's doing it on its own.

kevinkk525, 2018-03-27:
I'm glad it works now.

Calling rebooting in case of panic handler a fix for production readiness is quite funny though. It's a nice workaround until the firmware is stable but nothing more.

bdespatis, 2018-03-27:
I know, but when you need to ship now and fix later, it's much easier if the CPU does not halt. That is why I am so happy. Sometimes, it's the only viable solution.

bdespatis, 2018-03-27:
````
FreeRTOS running on BOTH CORES, MicroPython task started on App Core.
Running from OTA_0 partition starting at 0x10000, [MicroPython_1].

 Reset reason: Power on reset
    uPY stack: 64512 bytes
     uPY heap: 3001600/2064/2999536 bytes (in SPIRAM using heap_caps_malloc)

MicroPython XYZ - 2018-03-27 on XYZ with WROVER-32
Type "help()" for more information.
>>> XYZ IP: 192.168.0.39
Broker IP: 192.168.0.100
ID: XYZ_39
(0, 'Disconnected')
/[...]/[...]/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008e5fc on core 1
0x4008e5fc: bt_i2c_set_wifi_data at ??:?

Backtrace: 0x40092e64:0x3ffd14c0 0x4009305f:0x3ffd14e0 0x4008e5fc:0x3ffd1500 0x401897f6:0x3ffd1540 0x40175689:0x3ffd1560 0x40175d17:0x3ffd1590 0x401749cd:0x3ffd15b0
0x40092e64: ram_pbus_rx_dco_cal at ??:?

0x4009305f: ram_pbus_rx_dco_cal at ??:?

0x4008e5fc: bt_i2c_set_wifi_data at ??:?

0x401897f6: mbedtls_sha512_process at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/mbedtls/library/sha512.c:247 (discriminator 1)

0x40175689: create_name at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

0x40175d17: f_open at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

0x401749cd: find_volume at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

Rebooting...
````

At least now it reboots by itself.

I don't have any MicroPython on Core1 if I run on App Core, do I ?

If it is the case, should I point finger to another FreeRTOS Task that is running (WiFi, MQTT, Telnet) ? Next step before crash was MQTT Client Connect Task.

I don't want to fix this only by using one core for FreeRTOS. Thanks for the help !

kevinkk525, 2018-03-27:
It looks like you are connecting to a broker using ssl. If that's the case then loboris should know an answer.
You are using the mqtt library  of the fork?

bdespatis, 2018-03-27:
````
/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/freertos/./queue.c:719 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008e5fc on core 0
0x4008e5fc: bt_i2c_set_wifi_data at ??:?

Backtrace: 0x40092e64:0x3ffd14c0 0x4009305f:0x3ffd14e0 0x4008e5fc:0x3ffd1500 0x401897f6:0x3ffd1540 0x40175689:0x3ffd1560 0x40175d17:0x3ffd1590 0x401749cd:0x3ffd15b0
0x40092e64: ram_pbus_rx_dco_cal at ??:?

0x4009305f: ram_pbus_rx_dco_cal at ??:?

0x4008e5fc: bt_i2c_set_wifi_data at ??:?

0x401897f6: mbedtls_sha512_process at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/mbedtls/library/sha512.c:247 (discriminator 1)

0x40175689: create_name at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

0x40175d17: f_open at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

0x401749cd: find_volume at /home/pi/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/fatfs/src/ff.c:4730

Rebooting...
````

Now on Core0. I did not change the code since this morning. I clone again this repository and I used most of defaults config. I use SPIFFS too, but error is showing fatfs ...

Thanks Boris or Kevin in advance for all your help !

bdespatis, 2018-03-27:
I don't use SSL for MQTT, but I do use the one in this repository (FreeRTOS task).

kevinkk525, 2018-03-27:
please post a small code of how you import mqtt and how you connect to your broker. also attach your sdkconfig.

hargathor, 2018-03-27:
Hi, did you try publishing without subscribing ?

bdespatis, 2018-03-28:
I did not change my MicroPython Script and revert it to a previous commit. I will be back with code exemple to reproduce. I think we may have a problem with FreeRTOS MQTT task or my sdkconfig file ... Very strange.

loboris, 2018-05-17:
As the mqtt module was updated, and many changes which may be related to this issue were commited, I'll close this issue.
Please reopen if needed.

closed 2018-05-17 by loboris