loboris/MicroPython_ESP32_psRAM_LoBo #113: RTC SYNC not recognising IST

theKatBelgooly, 2018-03-28:
Hi
This issue is certainly not a show stopper for me, just annoying in that I need to know how things work :-) I have this string defined in my sdkconfig "GMT0IST,M3.5.0/1,M10.5.0" and have the following code in my main.py "rtc.ntp_sync(server="hr.pool.ntp.org")" however, the time is set to UTC and not UTC+1:

Synchronize time from NTP server ...
Time set: Wed Mar 28 08:42:26 2018    <--- current local time is actually 09:42

Please can I have the correct setting/string for IST? 

.... one other minor issue, please can you change the timestamp on the boot header, it's reflecting 2017 and not 2018 as in:

MicroPython ESP32_LoBo_v3.1.27 - 2017-03-21 on Riverbank Shed with ESP32

Keep up the good work :-)

loboris, 2018-03-29:
More info about this issue on the [forum](https://loboris.eu/forum/Thread-Timezone-and-NTP-Server-error-adjusting-time?pid=21#pid21)

I would suggest to use the simple time string "<UTC+offset>" or "<UTC-offset>".

loboris, 2018-04-08:
I'm not shure if I can do anything more about this issue at the moment.

Please reopen if needed.

closed 2018-04-08 by loboris