loboris/MicroPython_ESP32_psRAM_LoBo #114: ILI9341 with touch, SPI problem

jeffmer, 2018-03-28:
I have the display module working with a TTGO V1.0 ( 4M PSRAM) module driving a ILI9341 screen, using latest firmware 21-3-18

The display functions work fine, however, touch screen does not. If I deinit the display, I can get a micropython xpt2046 driver working with the touch screen, however, this does not work with the display initialised - neither display or screen works but generate no error reports. 

The display and touchscreen drivers share SPI(1) as they do in TFT. I suspect this is a problem with the SPI driver but can get no further. 

PS Keep up the good work, having a large heap is truly liberating.

loboris, 2018-04-08:
Sorry for late reply.
I couldn't reproduce the issue with displays I've tested.

Could you give more information about the display you are using and the initialization parameters?

flcorley, 2019-01-22:
I'm having the same problem. Or at least I think its the same. I didn't try a separate xpt2046 driver. My configuration works with the arduino ide bur no touchscreen in upython.
I'm using this [ILI9341](https://www.banggood.com/2_8-Inch-ILI9341-240x320-SPI-TFT-LCD-Display-Touch-Panel-SPI-Serial-Port-Module-p-1206782.html?rmmds=myorder&cur_warehouse=CN) display that has a xpt2046 touchscreen controller. I'm using an [ESP32](https://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html?cur_warehouse=CN) dev board.
This is what I'm using for an init line - 
`tft.init(tft.ILI9341, width=240, height=320, miso=19,mosi=23,clk=18,cs=5,dc=2,rst_pin=4,tcs=21,hastouch=True, bgr=True)`
I copied it from the examples, substituting my pin numbers (and adding reset). 

I also tried using  tft.TOUCH_XPT
`tft.init(tft.ILI9341, width=240, height=320, miso=19,mosi=23,clk=18,cs=5,dc=2,rst_pin=4,tcs=21,hastouch=tft.TOUCH_XPT, bgr=True)`

In both cases, the display works great but no touchscreen.

patbeppo, 2019-02-02:
I have apparently the same problem as jeffmer. I am using a WEMOS LOLIN D32 PRO with a 2.4" TFT, connected to the 10 wire connector on the board. The TFT is controlled by an ILI9341 and the touch screen by an XPT2064. (https://wiki.wemos.cc/products:d32:d32_pro)

The display works fine even with SPI speeds up to 80Mhz on VSPI. I used your tftdemo.py to test. 
The touch panel does not work in the demo. Also calling tft.gettouch() always returns False, 0, 0.
I tested SPI speed down to 100kHz: never worked. So speed do not appear to be the issue. Also the touchpanel calibration does not work: the screen comes up but it never reacts to my touches.

The touch panel does work if I use machine.SPI with the same pin configuration exept for **not giving the CS-pin for the touch panel**. I also noted that when running tftdemo.testt() that tp cs was high (measured with a multi meter; I do not have an osci...). 
If I set tp cs to low I can read from the touch panel by writing 0xb0 resp. 0xd0 resp. 0x90. Surprisingly tp cs does not need to be set high after sending the command byte. I can leave it low and keep reading...

From these observations I am susptecting that your code might not set tp cs low long enough or maybe not at all???  Could this have to do with routing of the tp cs pin?
I looked through your C-code, but that is mostly beyond my level. I noticed however the following:
In tftspi.c in touch_get_data you write that the touch device **must** already be selected.
However, in the first line of executed code you none the less select it. 

// get 16-bit data from touch controller for specified type
// ** Touch device must already be selected **
//----------------------------------------
int IRAM_ATTR touch_get_data(uint8_t type)
{
    /*
    esp_err_t ret;
    spi_lobo_transaction_t t;
    memset(&t, 0, sizeof(t));            //Zero out the transaction
    uint8_t rxdata[2] = {0};

    // send command byte & receive 2 byte response
    t.rxlength=8*2;
    t.rx_buffer=&rxdata;
    t.command = type;

    ret = spi_lobo_transfer_data(ts_spi, &t);    // Transmit using direct mode

    if (ret != ESP_OK) return -1;
    return (((int)(rxdata[0] << 8) | (int)(rxdata[1])) >> 4);
    */
    spi_lobo_device_select(ts_spi, 0);

    ts_spi->host->hw->data_buf[0] = type;
    _spi_transfer_start(ts_spi, 24, 24);
    uint16_t res = (uint16_t)(ts_spi->host->hw->data_buf[0] >> 8);

    spi_lobo_device_deselect(ts_spi);

    return res;
}

Here is my initialization:

tft.init(tft.ILI9341, spihost=tft.VSPI, speed= 80000000, width=240, height=320, miso=19, mosi=23, clk=18, dc=27, cs=14, rst_pin=33, backl_pin=32, backl_on=1, hastouch=tft.TOUCH_XPT, tcs=12, splash=False)

derkomai, 2019-03-18:
Could [this](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/178#issuecomment-440952721) be related? This guy reports he has a xpt2046 module working after swapping some bytes.