loboris/MicroPython_ESP32_psRAM_LoBo #115: Timezone incorrect for New Zealand, is GMT+12, not GMT-12

BluSpanner, 2018-03-30:
The timezone for New Zealand is incorrect.
New Zealand is GMT+12, not GMT-12

I have tested the following does resolve the issue.

modify MicroPython_BUILD/components/micropython/docs/zones.csv
-"Pacific/Auckland","NZST-12NZDT,M9.5.0,M4.1.0/3"
+"Pacific/Auckland","NZST+12NZDT,M9.5.0,M4.1.0/3"

Thanks

loboris, 2018-03-30:
Please check https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/113

More info about this issue on the [forum](https://loboris.eu/forum/Thread-Timezone-and-NTP-Server-error-adjusting-time?pid=21#pid21)

BluSpanner, 2018-03-31:
Understand there may be a bug in the esp-idf, but..
Other items in the zones.csv match the wikipedia data below - except entry for NZST
It's not consistent with entry for "New Zealand (most areas)" in https://en.wikipedia.org/wiki/List_of_tz_database_time_zones, which is +12

loboris, 2018-03-31:
Time zone file from [here](https://github.com/nayarsystems/posix_tz_db) is used.

The zone string is extracted from **/usr/share/zoneinfo/** with simple Python command:
```python
>>> import sys
>>> ZONES_DIR = "/usr/share/zoneinfo/"
>>> TIMEZONE = "Pacific/Auckland"
>>> open(ZONES_DIR + TIMEZONE, "rb").read().split(b"\n")[-2]
'NZST-12NZDT,M9.5.0,M4.1.0/3'
```
You can try it on your PC (if running on Linux) to test it.

Sorry, but I can't do anything about it.

loboris, 2018-04-08:
I'm not shure if I can do anything more about this issue at the moment.

Please reopen if needed.

closed 2018-04-08 by loboris