loboris/MicroPython_ESP32_psRAM_LoBo #117: SSL not working with some servers

carterw, 2018-03-31:
I've got a server on the internet with a good SSL certificate and it is configured properly as reported by online checkers. With previous versions of the LoBo firmware I could connect a socket to the server and then successfully "wrap" the socket in SSL whereupon the correct handshaking would take place.

With the most recent version of the LoBo firmware the SSL handshake fails when the "wrap" is attempted. I get an error code of 7200, which appears to be MBEDTLS_ERR_SSL_INVALID_RECORD.

Here you can see that a test program works with the older firmware;
```
>>> sys.implementation
(name='micropython', version=(0, 0, 1))
>>> import nt
Starting WiFi ...
I (11840) phy: phy_version: 380.0, b77d4dc, Jan 24 2018, 17:42:07, 0, 0
[(2, 1, 0, 'canulus.com', ('138.197.54.244', 443))]
('138.197.54.244', 443)
connected
wrapped
b'HTTP/1.1 200 OK\r\nX-Powered-By: Express\r\nContent-Type: text/html; charset=utf-8\r\nContent-Length: 170\r'
>>> 
```
And here you can see the same program running (and failing) on a recent build;
```
>>> sys.implementation
(name='micropython', version=(3, 1, 21))
>>> import nt
Starting WiFi ...
[(2, 1, 0, 'canulus.com', ('138.197.54.244', 443))]
('138.197.54.244', 443)
connected
mbedtls_ssl_handshake error: -7200
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "nt.py", line 75, in <module>
  File "nt.py", line 73, in run
  File "nt.py", line 31, in http_get
OSError: [Errno 5] EIO
```
The programs runs successfully in Python on a Linux machine as well, so there must be something about this current version of the firmware that has changed. Possible a micropython setting?

The error code is mentioned in the Espressif docs;
[http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html](url)

```
MBEDTLS_SSL_MAX_CONTENT_LEN

TLS maximum message content length

Found in: Component config > mbedTLS

Maximum TLS message length (in bytes) supported by mbedTLS.

16384 is the default and this value is required to comply fully with TLS standards.

However you can set a lower value in order to save RAM. This is safe if the other end of the connection supports Maximum Fragment Length Negotiation Extension (max_fragment_length, see RFC6066) or you know for certain that it will never send a message longer than a certain number of bytes.

If the value is set too low, symptoms are a failed TLS handshake or a return value of MBEDTLS_ERR_SSL_INVALID_RECORD (-0x7200).
```

I've attached a zipped version of the test program that illustrates the problem...
[nt.py.gz](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1865989/nt.py.gz)

carterw, 2018-04-01:
I tried reducing the max fragment size on my server to 512, did not help the problem.

loboris, 2018-04-02:
The servers you are testing obviously do not support _Maximum Fragment Length Negotiation Extension (max_fragment_length, see RFC6066)_, and you must set
`→ Component config → mbedTLS → TLS maximum message content length` in **menuconfig** to minimum of **16384** bytes.
You will have less memory for MicroPython and it may be necessary to set the MicroPython heap to 72 or 64 KB or use the board with psRAM.

When configured this way it works with your test program:

```python
>>> server = "canulus.com"
>>> http_get(server, port, path)
[(2, 1, 0, 'canulus.com', ('138.197.54.244', 443))]
('138.197.54.244', 443)
connected
wrapped
b'HTTP/1.1 200 OK\r\nServer: nginx/1.10.3 (Ubuntu)\r\nDate: Mon, 02 Apr 2018 19:45:33 GMT\r\nContent-Type: t'
>>> server = "docs.micropython.org"
>>> http_get(server, port, path)
[(2, 1, 0, 'docs.micropython.org', ('176.58.119.26', 443))]
('176.58.119.26', 443)
connected
wrapped
b'HTTP/1.1 302 Moved Temporarily\r\nServer: nginx/1.12.2\r\nDate: Mon, 02 Apr 2018 19:46:08 GMT\r\nContent-T'
>>> 
```

It also works with curl module (only with `TLS maximum message content length`=16384).

carterw, 2018-04-02:
Thanks for the info. I had installed Nginx on my server over the weekend as a reverse-proxy intermediary for back-end programs (as you can see - Server: nginx/1.10.3). It handshakes with the current firmware even without having to modify any settings.

Looking at the build menu however, I am seeing that the content length is already 16384 by default and this was failing with my previous server setup (a Node.js server program). That server code had worked properly with the previous firmware release, so something must have changed.

I can close this issue since I have a workaround, but are you sure the content length in the menu is what is being used by the code? If so there may be some other subtle issue that caused the problem. I could turn off Nginx and go to the previous mode if you are willing to take a quick look.

carterw, 2018-04-17:
Re-opening this, I am seeing the problem occur again. I even tried getting a new SSL certificate (but it is from the same vendor, it may be a large certificate). I attempted to reduce the SSL buffer size on Nginx down to 1K but no change, it may be that the fragment size restriction only takes effect after handshake.

I did some looking on the mbedtls github site, apparently this has been seen by others as well and the code owners claim it has been fixed. Are we using a recent implementation of mbedtls?

[https://github.com/ARMmbed/mbedtls/issues/387#issuecomment-381899090](url)

closed 2018-04-19 by carterw

lypanov, 2018-04-26:
Wouldn't it make sense for at least the PSRAM configurations to have this default to 16kb? It currently uses 4kb. After reconfiguring to 16kb I'm able to access an SSL API that I was previously unable to access. Maybe a new firmware configuration? (I know.. you already have many).