loboris/MicroPython_ESP32_psRAM_LoBo #118: Double Ctrl-D cause crash

cocoaway, 2018-04-01:
Compiled on macOS, MicroPython ESP32_LoBo_v3.1.29 - 2018-03-31 on ESP32 board with ESP32

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57
 
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4704
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13204
entry 0x40078d44

Internal FS (LittleFS ver 1.4): [on top of ESP32 wear leveling]
Mounted on partition 'internalfs' [size: 2818048; Flash address: 0x150000]

Filesystem size: 2777088 B
           Used: 2560 B
           Free: 2774528 B

 
FreeRTOS running on BOTH CORES, MicroPython task started on App Core.
Running from Factory partition starting at 0x10000, [MicroPython].
 
 Reset reason: RTC WDT reset
    uPY stack: 19456 bytes
     uPY heap: 80000/6512/73488 bytes
 
MicroPython ESP32_LoBo_v3.1.29 - 2018-03-31 on ESP32 board with ESP32

**Connect to serial, then Ctrl-D and again Ctrl-D.**

Result : 

**### Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)**
. Exception was unhandled.
Core 1 register dump:
PC      : 0x3ffc1ead  PS      : 0x00060930  A0      : 0x80120ce1  A1      : 0x3ffb8150  
A2      : 0x3ffc1cb8  A3      : 0x3ffc1ead  A4      : 0x000002a9  A5      : 0x00005000  
A6      : 0xffffffff  A7      : 0x00000003  A8      : 0x80120a1e  A9      : 0x3ffb8110  
A10     : 0x3ffc1cac  A11     : 0x00004000  A12     : 0x00001000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffc1cb5  SAR     : 0x00000006  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3ffc1eac  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
 
Backtrace: 0x3ffc1ead:0x3ffb8150 0x40120cde:0x3ffb8180 0x4011f9ab:0x3ffb81a0 0x40100921:0x3ffb81c0 0x400f9fca:0x3ffb81e0 0x400d5715:0x3ffb8200 0x400d4737:0x3ffb
8220
 
CPU halted.

loboris, 2018-04-02:
Thank you for reporting, I'll check it.

loboris, 2018-04-05:
The behaviour of the `Ctrl+D` was changed, it now performs full ESP32 soft reset.
There should be no issues with it anymore.

closed 2018-04-05 by loboris