loboris/MicroPython_ESP32_psRAM_LoBo #119: Usocket error

Lisa999, 2018-04-04:
```
>>> import usocket as _socket
>>> ai = _socket.getaddrinfo("www.micropython.org", 80, 0, _socket.SOCK_STREAM)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function takes 2 positional arguments but 4 were given
``` 

Conclusion: not compatible with standard micropython version

mkarliner, 2018-04-04:
I went through this joy a while back.
May I suggest that Lobo is based on the official repos where ever possible and clearly states when it isn't in the README or other relevant place?
It seems to me that the governance of micropython in general is subject to a lot of ego conflicts which has put me off the project for long periods.

Lisa999, 2018-04-04:
`May I suggest that Lobo is based on the official repos where ever possible and clearly states when it isn't in the README or other relevant place?`
I second that! It would save me the time to find out why my app isn't working...

Lisa999, 2018-04-04:
And there are more socket difference i'm noticing:
socket.setblocking(False) should be the same as socket.settimeout(0) according to the docs. But it isn't, socket.setblocking(False) is giving an OSError: [Errno 110] ETIMEDOUT when doing a socket.accept()
Not good...

loboris, 2018-04-05:
**usocket** module is updated and `getaddrinfo` accepts additional arguments (which are ignored, same as in official MicroPython).

@Lisa999 
`socket.setblocking()` and `socket.accept()` behaviour is **exactly the same** as in official MicroPython.
`socket.accepted()` method is added which is the same as `socket.accept()` but does not raise an exception.

The last MicroPython core synchronization to the official MicroPython repository was to the commit **bcfff4f** (2018/03/30).

The list of all modified sources is available [here](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/README.md).
The modifications to the sources are mostly minor changes to make them work on this port, with only a couple of exceptions.

Lisa999, 2018-04-06:
socket.accepted() method is added which is the same as socket.accept() but does not raise an exception.
accepted() doesn't support setblocking nor streams... ;-(

loboris, 2018-04-08:
I think it is clear that the `usocket` module is the same as official MicroPython's one, so I'll close this issue.

If there are some issues with any of the module's methods or functionality, maybe it would be better to open an issue at official MicroPython's repository.

Please reopen if needed.

closed 2018-04-08 by loboris