loboris/MicroPython_ESP32_psRAM_LoBo #120: Error when attempting to connect over USB with rshell

stroobandt, 2018-04-05:
After flashing the prebuilt firmware `MicroPython_LoBo_esp32_all` dated 5 April 2018 on a new ESP-WEMOS-32 and rebooting, any attempt to connect using `rshell` fails with the following extensive error:
```
$ rshell -p /dev/ttyUSB2 -b 115200

Connecting to /dev/ttyUSB2 ...
b'OK\r\nets Jun  8 2016 00:22:57\r\n\r\nrst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)\r\nconfigsip: 0, SPIWP:0xee\r\nclk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00\r\nmode:DIO, clock div:2\r\nload:0x3fff0018,len:4\r\nload:0x3fff001c,len:5912\r\nload:0x40078000,len:0\r\nho 12 tail 0 room 4\r\nload:0x40078000,len:14028\r\nentry 0x4007901c\r\n\x1b[0;32mI (30) boot: ESP-IDF v3.1-dev-644-g139d4989 2nd stage bootloader\x1b[0m\r\n\x1b[0;32mI (30) boot: compile time 10:32:27\x1b[0m\r\n\x1b[0;32mI (31) boot: Enabling RNG early entropy source...\x1b[0m\r\n\x1b[0;32mI (37) boot: SPI Speed      : 40MHz\x1b[0m\r\n\x1b[0;32mI (41) boot: SPI Mode       : DIO\x1b[0m\r\n\x1b[0;32mI (45) boot: SPI Flash Size : 4MB\x1b[0m\r\n\x1b[0;32mI (49) boot: Partition Table:\x1b[0m\r\n\x1b[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length\x1b[0m\r\n\x1b[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000\x1b[0m\r\n\x1b[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000\x1b[0m\r\n\x1b[0;32mI (75) boot:  2 MicroPython      factory app      00 00 00010000 001c0000\x1b[0m\r\n\x1b[0;32mI (82) boot:  3 internalfs       Unknown data     01 82 001d0000 00100000\x1b[0m\r\n\x1b[0;32mI (90) boot: End of partition table\x1b[0m\r\n\x1b[0;32mI (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x687b8 (427960) map\x1b[0m\r\n\x1b[0;32mI (252) esp_image: segment 1: paddr=0x000787e0 vaddr=0x3ffb0000 size=0x07830 ( 30768) load\x1b[0m\r\n\x1b[0;32mI (265) esp_image: segment 2: paddr=0x00080018 vaddr=0x400d0018 size=0x131c7c (1252476) map\x1b[0m\r\n\x1b[0;32mI (704) esp_image: segment 3: paddr=0x001b1c9c vaddr=0x3ffb7830 size=0x02960 ( 10592) load\x1b[0m\r\n\x1b[0;32mI (708) esp_image: segment 4: paddr=0x001b4604 vaddr=0x40080000 size=0x00400 (  1024) load\x1b[0m\r\n\x1b[0;32mI (711) esp_image: segment 5: paddr=0x001b4a0c vaddr=0x40080400 size=0x13860 ( 79968) load\x1b[0m\r\n\x1b[0;32mI (752) esp_image: segment 6: paddr=0x001c8274 vaddr=0x400c0000 size=0x0064c (  1612) load\x1b[0m\r\n\x1b[0;32mI (753) esp_image: segment 7: paddr=0x001c88c8 vaddr=0x50000000 size=0x00984 (  2436) load\x1b[0m\r\n\x1b[0;32mI (775) boot: Loaded app from partition at offset 0x10000\x1b[0m\r\n\x1b[0;32mI (775) boot: Disabling RNG early entropy source...\x1b[0m\r\n\x1b[0;32mI (776) cpu_start: Pro cpu up.\x1b[0m\r\n\x1b[0;32mI (779) cpu_start: Starting app cpu, entry point is 0x40081278\x1b[0m\r\n\x1b[0;32mI (770) cpu_start: App cpu up.\x1b[0m\r\n\x1b[0;32mI (790) heap_init: Initializing. RAM available for dynamic allocation:\x1b[0m\r\nD (797) heap_init: New heap initialised at 0x3ffae6e0\x1b[0m\r\n\x1b[0;32mI (802) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\x1b[0m\r\nD (808) heap_init: New heap initialised at 0x3ffc1fb0\x1b[0m\r\n\x1b[0;32mI (813) heap_init: At 3FFC1FB0 len 0001E050 (120 KiB): DRAM\x1b[0m\r\n\x1b[0;32mI (819) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\x1b[0m\r\n\x1b[0;32mI (825) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\x1b[0m\r\nD (832) heap_init: New heap initialised at 0x40093c60\x1b[0m\r\n\x1b[0;32mI (837) heap_init: At 40093C60 len 0000C3A0 (48 KiB): IRAM\x1b[0m\r\n\x1b[0;32mI (843) cpu_start: Pro cpu start user code\x1b[0m\r\nD (855) clk: RTC_SLOW_CLK calibration value: 3521088\x1b[0m\r\n\x1b[0;32mI (195) cpu_start: Starting scheduler on PRO CPU.\x1b[0m\r\n\x1b[0;32mI (0) cpu_start: Starting scheduler on APP CPU.\x1b[0m\r\nD (196) heap_init: New heap initialised at 0x3ffe0440\x1b[0m\r\nD (200) heap_init: New heap initialised at 0x3ffe4350\x1b[0m\r\n\x1b[0;32mI (205) MicroPython: [=== Started ===]\x1b[0m\r\n\r\nInternal FS (SPIFFS): Mounted on partition \'internalfs\' [size: 1048576; Flash address: 0x1D0000]\r\n----------------\r\nFilesystem size: 956416 B\r\n           Used: 512 B\r\n           Free: 955904 B\r\n----------------\r\n\r\nFreeRTOS running on BOTH CORES, MicroPython task started on App Core.\r\nRunning from Factory partition starting at 0x10000, [MicroPython].\r\n\r\n Reset reason: Soft CPU reset\r\n    uPY stack: 19456 bytes\r\n     uPY heap: 72000/6112/65888 bytes\r\n\r\nMicroPython ESP32_LoBo_v3.2.2 - 2018-04-04 on ESP32 board with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1185, in connect
    ip_address = socket.gethostbyname(port)
socket.gaierror: [Errno -2] Name or service not known

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/rshell", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/rshell/command_line.py", line 4, in main
    rshell.main.main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2658, in main
    real_main()
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 2620, in real_main
    connect(args.port, baud=args.baud, wait=args.wait, user=args.user, password=args.password)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1191, in connect
    connect_serial(port, baud=baud, wait=wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1215, in connect_serial
    dev = DeviceSerial(port, baud, wait)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1436, in __init__
    Device.__init__(self, pyb)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1262, in __init__
    self.has_buffer = self.remote_eval(test_buffer)
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1353, in remote_eval
    return eval(self.remote(func, *args, **kwargs))
  File "/usr/local/lib/python3.5/dist-packages/rshell/main.py", line 1331, in remote
    self.pyb.enter_raw_repl()
  File "/usr/local/lib/python3.5/dist-packages/rshell/pyboard.py", line 193, in enter_raw_repl
    raise PyboardError('could not enter raw repl')
rshell.pyboard.PyboardError: could not enter raw repl
```

stroobandt, 2018-04-05:
```
$ dmesg
…
[ 4850.975891] usb 5-2: cp210x converter now attached to ttyUSB2
```
Notwithstanding, connecting through `ampy` remains without any response:
```
$ ampy --port /dev/ttyUSB2 ls
```

loboris, 2018-04-05:
Thanks for reporting and sorry for the bug.

It is fixed now and commited. Please test.

stroobandt, 2018-04-05:
That works now, thanks Boris!
The REPL announces:

    MicroPython ESP32_LoBo_v3.2.3 - 2018-04-04 on ESP32 board with ESP32

I am not sure if that date is intentional.

Also, [the pertaining Wiki firmware download page](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/firmwares) looks like nothing has changed since January, which of course is not true. One needs to dig deep into the `.zip` file to find a date that shows that one is dealing with a far more recent firmware. Perhaps, you should think of a script that automatically updates the firmware date on the Github wiki.

loboris, 2018-04-05:
You are right, I forgot to update the date.

Wiki firmware download page was edited in January, but the links always points to the latest firmwares.

stroobandt, 2018-04-05:
> Wiki firmware download page was edited in January, but the links always points to the latest firmwares.

Yes, I have no doubt about that. However, the casual visitor receives no visual clue that new firmware has been made available on a very frequent basis. One might leave with the impression that nothing has changes since January.

kevinkk525, 2018-04-05:
I guess that's no problem once loboris uses the release functionality of github to publish the compiled firmware versions.

stroobandt, 2018-04-05:
Sorry, I have to reopen this issue. `ampy` still does not work.

    $ ampy --port /dev/ttyUSB2 put ../main.py

loboris, 2018-04-05:
Working here:

```
boris@UbuntuMate:~$ ampy --port /dev/ttyUSB0 ls
flash
sd
boris@UbuntuMate:~$ ampy --port /dev/ttyUSB0 ls /flash
boot.py
test1.jpg
tiger1.bmp
xxx.sh
boris@UbuntuMate:~$ ampy --port /dev/ttyUSB0 ls /sd/tft
tftdemo.py
paint.py
font_tool
fonts
images
boris@UbuntuMate:~$ 
boris@UbuntuMate:~$ ampy --port /dev/ttyUSB0 get /flash/boot.py
# This file is executed on every boot (including wake-boot from deepsleep)
import sys
sys.path[1] = '/flash/lib'

boris@UbuntuMate:~$ 

```

meigrafd, 2018-04-05:
Works for me too after last Update:
```
root@raspberrypi:~# ampy --port /dev/ttyUSB0 ls
flash
sd
root@raspberrypi:~# rshell -p /dev/ttyUSB0 -b 115200
Connecting to /dev/ttyUSB0 ...
Welcome to rshell. Use Control-D to exit.
/root> repl
Entering REPL. Use Control-X to exit.
>
MicroPython ESP32_LoBo_v3.2.3 - 2018-04-04 on ESP32 board with ESP32
Type "help()" for more information.
>>> 
>>> 
```

loboris, 2018-04-08:
I'll close the issue as it looks solved.

Please reopen if needed.

closed 2018-04-08 by loboris

stroobandt, 2018-04-08:
I tested again with today's firmware of 2018-04-08.
Both `rshell` and `ampy` work fine now on a Heltec LoRa.