loboris/MicroPython_ESP32_psRAM_LoBo #122: Attempting to read ESP32 internal Temperature sensor results in a crash

BluSpanner, 2018-04-07:
```
>>> import machine
>>> adc = machine.ADC(machine.Pin(35))
>>> adc.atten(machine.ADC.ATTN_11DB)
>>> adc.read()
assertion "(adc_num == ADC_UNIT_1) || (adc_num == ADC_UNIT_2)" failed: file "/home/LoBo2_Razno/ESP32/MicroPython/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp_adc_cal/./esp_adc_cal.c", line 303, function: esp_adc_cal_characterize
abort() was called at PC 0x40118e53 on core 1

Backtrace: 0x4009132c:0x3ffb8260 0x400914cf:0x3ffb8280 0x40118e53:0x3ffb82a0 0x401a3aa1:0x3ffb82d0 0x400fd392:0x3ffb8300 0x400ecd39:0x3ffb8350 0x400e86f5:0x3ffb8370 0x400e8765:0x3ffb8390 0x400f5896:0x3ffb83b0 0x400ece96:0x3ffb8450 0x400e86f5:0x3ffb84b0 0x400e8722:0x3ffb84d0 0x400dba6f:0x3ffb84f0 0x400dbcf1:0x3ffb85a0 0x400d53a0:0x3ffb85e0

CPU halted.
```

loboris, 2018-04-07:
@BluSpanner 
The **bug** in `adc.read()` was introduced couple of updates ago after updating esp-idf, sorry.
It is fixed now an will be commited today.
Thank you for reporting.

The method for reading internal temperature sensor is `machine.internal_temp()`
```python
>>> import machine
>>> machine.internal_temp()
(134, 56.66666793823243)
```
Wiki updated.

loboris, 2018-04-08:
The ADC `read()` fixed and commited.

closed 2018-04-08 by loboris