loboris/MicroPython_ESP32_psRAM_LoBo #124: network.WLAN can't work on WIFI_MODE_APSTA mode

0x1abin, 2018-04-13:
Hi
The latest version can't work on WiFi station + soft-AP mode:
```python
>>> import network
>>> wlan_sta = network.WLAN(network.STA_IF)
>>> wlan_sta.active(True)
True
>>> wlan_ap = network.WLAN(network.AP_IF)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: WiFi already started in different mode
>>> 

```

kevinkk525, 2018-04-13:
It did not work correctly with previous versions too.
To prevent people from using it, loboris made that exception temporarily until he fixes the problems with WIFI_MODE_APSTA.

loboris, 2018-04-14:
I've been working on updating the **network** module.
STA, AP and APSTA modes now works as expected, for example in APSTA mode ftp server can be connected on both STA and AP interfaces (only one connection is supported, so not at the same time) and he web server (microWebServer) can be accessed on both interfaces at the same time.

I'm doing some final tests and the update will be committed later today or tomorrow.

loboris, 2018-04-15:
The update was commited and operating in APSTA mode is fully supported.

closed 2018-04-15 by loboris