loboris/MicroPython_ESP32_psRAM_LoBo #127: Inverted UART TX please

larsmm, 2018-04-15:
I need an invertes UART TX for sending 38 kHz modulated IR signals. The ESP32 hardware has a register for that: "The outgoing bit stream can be inverted by appropriately configuring the register UART_TXD_INV."

Please add this function.

loboris, 2018-04-15:
Thanks for the suggestion, I'll try to implement it.

BTW, the module for IR controll is planned and will be implemented using ESP32 [RMT module](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/rmt.html) which is designed for that purpose.

larsmm, 2018-04-17:
Both functions would be very helpful! Thanks! :D

Here some help: all UART data pins can be inverted separately in UART_CONF0_REG register. Or maybe add an option to directly set the bits of this register. There are many more options, maybe people will need them.

loboris, 2018-04-17:
Inverting UART pins will be added on the next commit.
All combinations of inverting tx, rx, cts and rts will be available as defined in [esp-idf](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/uart.html#_CPPv221uart_set_line_inverse11uart_port_t8uint32_t).

loboris, 2018-04-18:
Setting the inverted pins is implemented in latest commit.
The Wiki page will be updated soon.

closed 2018-04-18 by loboris

larsmm, 2018-04-18:
Thank you very much! :D