loboris/MicroPython_ESP32_psRAM_LoBo #128: non ascii characters in uart callback pattern

fabianWhoknows, 2018-04-15:
Hello @loboris 
Thank you for this very cool project!
I have an issue with the callback function of the machine.UART module:
I need to add a pattern that does contain non ASCII characters. The pattern is the following b'\xff\xff\xff'.
Unfortunately it gets converted into some weird string:
```
>>> uart.callback(uart.CBTYPE_PATTERN, cb, pattern = '\xff\xff\xff')
>>> uart
UART(1, baudrate=768307, bits=3, parity=None, stop=1, tx=17, rx=5, rts=-1, cts=-1, timeout=0, buf_size=512, lineend=b'\xc3\xbf')
     pattern CB: True, pattern: [ÿÿÿ]
     Event task minimum free stack: 612
```

when I set the pattern in the bytes type it is ignored..
```
uart.callback(uart.CBTYPE_PATTERN, cb, pattern = b'\xff\xff\xff')
>>> uart
UART(1, baudrate=768307, bits=3, parity=None, stop=1, tx=17, rx=5, rts=-1, cts=-1, timeout=0, buf_size=512, lineend=b'\xc3\xbf')
     pattern CB: True, pattern: []
     Event task minimum free stack: 612
```
Can you tell me how I can set the pattern?

loboris, 2018-04-15:
The callback pattern is handled as string and only characters **\x00** - **\0x7f** are accepted correctly.

I'll change this so that any string or bytearray can be accepted (**\x00** - **\xff**).
I'll try to commit the update tomorrow.

fabianWhoknows, 2018-04-16:
@loboris:
This is what I suspected.. I think there is the same issue with the lineend argument in the uart.init() function.. Maybe one can change this in one go.

Thank you very much for the fast response and the whole project in general!

loboris, 2018-04-16:
The update is commited.

If the `lineend` or `pattern` contains characters > **\x7f**, the argument must be entered as bytearray (`b'\xff\xfe\x01\x41'`)
```python
MicroPython ESP32_LoBo_v3.2.11 - 2018-04-16 on ESP32 board with ESP32
Type "help()" for more information.
>>> import machine
>>> uart = machine.UART(2, rx=25, tx=26, baudrate=9600, bits=8, parity=None, stop=1, timeout=200, buffer_size=256, lineend=b'\xfe\xff')
I (30034) uart: ALREADY NULL
I (30034) uart: queue free spaces: 10
>>> uart
UART(2, baudrate=9603, bits=8, parity=None, stop=1, tx=26, rx=25, rts=-1, cts=-1, timeout=200, buf_size=512, lineend=b'\xfe\xff')
     Event task minimum free stack: 488
>>> def uart_cb(res):
...     print("[UART CB]", res)
... 
>>> uart.callback(uart.CBTYPE_PATTERN, uart_cb, pattern=b'\xff\xfe\x01\x41')
>>> uart
UART(2, baudrate=9603, bits=8, parity=None, stop=1, tx=26, rx=25, rts=-1, cts=-1, timeout=200, buf_size=512, lineend=b'\xfe\xff')
     pattern CB: True, pattern: b'\xff\xfe\x01A'
     Event task minimum free stack: 456
>>> 
OK.

```

fabianWhoknows, 2018-04-16:
@loboris 
Thank you for the fast update. I tested the pattern function as well as the lineend with some none ASCII pattern and it works! However I might have spotted something else:
With the following code
```
>>> def cb(res):
...     print(res)
>>> uart.callback(uart.CBTYPE_PATTERN, cb, pattern=b'\xff\xff\xff')
```
I get the following from the callback:
`(1,2,4)` when sending `b'e\x00\x07\x01\xff\xff\xff''` and `(1,2,10)` when sending `b'pI1: --- A\xff\xff\xff` through UART1.. So I suspect that the 3rd entry in the return tuple contains the number of bytes of the message (minus the number of bytes in the pattern of course) rather then `received_string` ..

Using the `uart.CBTYPE_DATA` gives me the same effect..

larsmm, 2018-04-17:
Same problem here on version 3.2.11 - 2018-04-16 with rx and tx jumpered together:
```
>>> from machine import UART
>>> uart = UART(1, tx=13, rx=12, timeout=5000, buffer_size=1024, baudrate=9600)
>>> def rx_event(res):
...     print('RX:', res)
...
>>> uart.callback(uart.CBTYPE_DATA, func=rx_event, data_len=10)
>>> uart.write('a1642stgsrcgsgdfhcsdgsgyfdvppppphf36')
36
>>> RX: (1, 1, 10)
RX: (1, 1, 10)
RX: (1, 1, 10)
uart.write('a1642stgsrcgsgdfhcsdgsgyfdvppppphf36')
36
>>> RX: (1, 1, 10)
RX: (1, 1, 10)
RX: (1, 1, 10)
RX: (1, 1, 10)
uart.write('a1642stgsrcgsgdfhcsdgsgyfdvppppphf36')
36
>>> RX: (1, 1, 10)
RX: (1, 1, 10)
RX: (1, 1, 10)
```

loboris, 2018-04-17:
I'm sorry, I've used the wrong callback argument type for the 3rd argument in callback function and the length was returned instead of the acctual data.
The update will be commited tomerrow.

loboris, 2018-04-18:
The UART module is updated and the callbacks are now working as expected.

closed 2018-04-18 by loboris

fabianWhoknows, 2018-04-18:
@loboris 
Thanks for the fix! Now it works like a charm!