loboris/MicroPython_ESP32_psRAM_LoBo #130: rtc - ntp with tz handling leads to error

newmindJP, 2018-04-20:
after connecting to WiFi I try following code:

from machine import RTC
rtc = RTC()
rtc.ntp_sync(server="192.168.1.10", tz="CET-1CEST-2")

This leads to error -> ValueError: tz string length must be 3 - 64
I tried several options, but when ever I set a tz it leads to this error

after checking the code in file "MicroPython_BUILD/components/micropython/esp32/machine_rtc.c" I guess the problems relates to line 333
`if ((strlen(tzs) < 2) && (strlen(tzs) < 64)) {`
the correct code has to be
`if ((strlen(tzs) > 2) && (strlen(tzs) < 64)) {`

loboris, 2018-04-21:
Yes it is a bug. Thank you for noticing that.
I'll fix it in the next update.

Until fixed, you can set the ime zone with
`sys.tz("CET-1CEST-2")`
and use `rtc.ntp_sync(server="192.168.1.10")` without setting the time zone.

That way the time zone is saved to NVS and will be restored after reset, power off or deepsleep wake up, so you have to set it only once.

loboris, 2018-05-14:
It was fixed in the latest update.

closed 2018-05-14 by loboris