loboris/MicroPython_ESP32_psRAM_LoBo #132: interrupt callback fires multiple times

untitledlt, 2018-04-23:
Here is my code:

```python
def cb(p):
    print(p.value())

pin = Pin(23, Pin.IN, Pin.PULL_DOWN)
pin.irq(trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING, handler=cb)
```

and here is output when i shorten the pin 23 to 3.3V:

```
1
1
1
1
1
1
0
0
0
0
0
0
0
1
1
1
1
1
0
0
0
0
0
0
0
0
```

Correct me if i'm wrong, but i think that callback should be called each time `pin.value()` changes so each line  should be the opposite of the previous one.
Why am i getting multiple `1` or `0` in a row?

nitruxa, 2018-04-23:
+1 same in here

loboris, 2018-05-17:
Sorry for the late reply.
Multiple callbacks are caused by the [contact bounce](https://www.allaboutcircuits.com/technical-articles/switch-bounce-how-to-deal-with-it/).

You can either debounce the pin with hardware, or use the debounce in callback function like in this simple example:
```python
from machine import Pin
import utime

intTime = 0
debTime = 100

def cb(p):
    global intTime
    if utime.ticks_diff(utime.ticks_ms(), intTime) > debTime:
        print(p.value())
        intTime = utime.ticks_ms()

pin = Pin(23, Pin.IN, Pin.PULL_DOWN)
pin.irq(trigger=Pin.IRQ_RISING | Pin.IRQ_FALLING, handler=cb)
```

BTW, you can use `Pin.ANYEDGE` instead of `Pin.IRQ_RISING | Pin.IRQ_FALLING`

I'm also planning to add the debouncing option to the Pin interrupt, so that it can be set as:
`pin.irq(trigger=Pin.IRQ_RISING, handler=cb, debounce=100)`

untitledlt, 2018-05-17:
@loboris if it was because of contact bouncing, pin value should differ each time but i'm getting multiple `0` or multiple `1` in a row.

loboris, 2018-05-17:
The bouncing is usually to short to be registered by the callback function.
For example:
- the pin state is low
- you connect the pin to 3.3V
- interrupt occurs and the callback function execution is requested
- contact bounces to low for a short period, the second callback function is requested, probably before the first one is even executed
- contact goes to high again and the third callback function is requested

At the time you are reading the pin value in the callback function, the pin state has already returned to high level.

The _Python callback function_ **cannot** be executed from the **ESP32 hw interrupt function**, the execution of the Python callback function is only **scheduled** to be executed as soon as possible.

The interrupt responses are very fast, and the execution of the **Python** function is slow (compared to how fast the interrupts can occur).

untitledlt, 2018-05-17:
Yeah, it makes sense. But maybe callback function could hold a pin state as it was while trigger has happened? It passes pin rederence to callback but we do not know it's real state as pin.value() returns current state.

loboris, 2018-05-17:
It should be easy to implement.
The callback function could receive the tuple argument (pin_id, value) where the value would be the pin state at the time of interrupt, so that your callback function can return the pin state at the time of the interrupt.
```python
def cb(res):
    print("Pin={}, value={}".format(res[0], res[1]))
```
The tuple argument could also contain the timestamp in micro seconds, so it would be possible to see the actual time between the interrupts.

I'll try to implement it, keeping the compatibility with the current behavior.

untitledlt, 2018-05-17:
Sounds good. Thanks!