loboris/MicroPython_ESP32_psRAM_LoBo #134: Neopixel module BUG

yoooman, 2018-04-28:
When I create two instances, I can't set color in normal.
The following is my test code.
Thank you.
```python
import machine
np = machine.Neopixel(22, 115, 0)
np2 = machine.Neopixel(26, 10, 0)
np.set(1, np.RED)
np.set(2, np.GREEN)
```

```bash
>>> import machine
>>> np = machine.Neopixel(22, 115, 0)
>>> np2 = machine.Neopixel(26, 10, 0)
>>> 
>>> np.set(1, np.RED)
>>> np.set(2, np.GREEN)

... in loop not return...
```

mzdaniel, 2018-05-14:
Issue confirmed in latest MicroPython ESP32_LoBo_v3.2.15 - 2018-05-10 on ESP32
Seems that the firmware cannot handle more than one neopixel GPIO.

This works displaying a green neopixel:

```python
import machine
np = machine.Neopixel(12, 10, 1)
np.set(1, np.GREEN)
```
Adding a new neopixel strip doesnt show the green one anymore:

```python
import machine
np = machine.Neopixel(12, 10, 1)
np2 = machine.Neopixel(14, 12, 0)
np.set(1, np.GREEN)
```

oldfartradioman, 2018-05-16:
nps = [machine.Neopixel(23,40,0), machine.Neopixel(22,40,0), machine.Neopixel(21,40,0),machine.Neopixel(19,40,0),machine.Neopixel(18,40,0)]
[i.setHSB( 10, .5, .5, .3  ) for i in nps]

I suppose that explains why this does not work.

loboris, 2018-05-19:
I have (finally) fixed the issue with multiple neopixels (on more than one gpio). The update will be commited in a couple of days.