loboris/MicroPython_ESP32_psRAM_LoBo #135: UART RX problems

larsmm, 2018-04-30:
A TSOP IR reciever is connected to the RX pin. Another piece of hardware is sending IR UART. So timing is not perfect, bytes get dropped or wrong bytes are recieved. After a few seconds or minutes I get one of the following errors.

Firmware: MicroPython ESP32_LoBo_v3.2.12 - 2018-04-18
Board: lolin32 (2 boards tested)

Program 1:
```python
from machine import UART

def uart_rx_event(res):
    print(res)

uart = UART(1, tx=13, rx=12, timeout=5000, buffer_size=32, baudrate=1200)
uart.callback(uart.CBTYPE_DATA, func=uart_rx_event, data_len=1)
```

Program 2:
```python
from machine import UART

uart = UART(1, tx=13, rx=12, timeout=5000, buffer_size=512, baudrate=1200)

while(True):
    if uart.any():
        print(uart.read(uart.any()))
```

Error (Program 1):
```
(1, 1, '!')
(1, 1, '?')
(1, 1, 'k')
(1, 1, '\x02')
(1, 1, '!')
(1, 1, '?')
(1, 1, 'k')
(1, 1, '\x02')
(1, 1, '!')
(1, 1, '?')
(1, 1, 'k')
(1, 1, '\x02')
(1, 1, '!')
(1, 1, '?')
(1, 1, 'k')
(1, 1, '\x02')
(1, 1, '!')
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x2f656d6f  PS      : 0x00060730  A0      : 0x800f68fb  A1      : 0x3ffc54d0
A2      : 0x3ffc54fc  A3      : 0x3ffc7b10  A4      : 0x00000000  A5      : 0x00000003
A6      : 0x3f4161f8  A7      : 0x3f42d924  A8      : 0x800edb09  A9      : 0x3ffc54b0
A10     : 0x3ffc54fc  A11     : 0x3ffc7b10  A12     : 0x00000000  A13     : 0xffffffff
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000000c  EXCCAUSE: 0x00000014
EXCVADDR: 0x2f656d6c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff

Backtrace: 0x2f656d6f:0x3ffc54d0 0x400f68f8:0x3ffc54f0 0x400f05a2:0x3ffc5530 0x400ebe91:0x3ffc5560 0x400f90cd:0x3ffc5580 0x400f0632:0x3ffc5620 0x400ebe91:0x3ffc5680 0x400ebed5:0x3ffc56a0 0x400ece70:0x3ffc56d0 0x400ed34d:0x3ffc5750 0x400ed3a7:0x3ffc5780 0x400d5a63:0x3ffc57a0 0x400def8c:0x3ffc57d0 0x400df303:0x3ffc57f0 0x400d542c:0x3ffc5830

CPU halted.
```

Error (Program 2):
```
b'\xf6'
b'!'
b'?'
b'k'
b'\xf2'
b'!'
b'\xbf'
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: mp_task
b'!'
b'?'
b'k'
b'\xf2'
b'\xef'
b'!'
b'?'
```

diginfo, 2018-05-04:
I also get "CPU Halted" frequently when trying to use the UART as well as some strange things that happen when using the callbacks.

This happened when the device was idle, nothing was being sent or received:

```
>>> assertion "time_after_timebase_us > s_timer_us_per_overflow" failed: file "/nfs/qnap/data/uPython/lobo/build/MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/esp32/./esp_timer_esp32.c", line 223, function: esp_timer_impl_set_alarm
abort() was called at PC 0x40110a1b on core 0

Backtrace: 0x40091438:0x3ffdbcb0 0x400915db:0x3ffdbcd0 0x40110a1b:0x3ffdbcf0 0x400820db:0x3ffdbd20 0x40081912:0x3ffdbd40 0x400819c9:0x3ffdbd60 0x4008cf90:0x3ffdbd80 0x4015768e:0x3ffdbda0 0x40159911:0x3ffdbdc0 0x4015992e:0x3ffdbde0 0x40159567:0x3ffdbe00 0x401597de:0x3ffdbe20 0x40089be0:0x3ffdbe40

CPU halted.
```

alankaranak, 2018-05-04:
I also have frequent errors when using UART

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x401a5653  PS      : 0x00060e30  A0      : 0x800f3f1b  A1      : 0x3ffc45e0
A2      : 0x00000001  A3      : 0x3f40fa70  A4      : 0x3ffc72a0  A5      : 0x3ffd7ad0
A6      : 0x3ffc4690  A7      : 0x3ffc45f0  A8      : 0x3f4122ec  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000019  A12     : 0x00000004  A13     : 0x3ffd7bbc
A14     : 0x3ffd7ab0  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x401a5653:0x3ffc45e0 0x400f3f18:0x3ffc4600 0x400eb05a:0x3ffc46a0 0x400e68b9:0x3ffc46d0 0x400e6929:0x3ffc46f0 0x400f3a3a:0x3ffc4710 0x400eb05a:0x3ffc47b0 0x400e68b9:0x3ffc47e0 0x400e6929:0x3ffc4800 0x400f3a3a:0x3ffc4820 0x400eb05a:0x3ffc48c0 0x400e68b9:0x3ffc48f0 0x400e6929:0x3ffc4910 0x400efd4b:0x3ffc4930 0x400ef3bf:0x3ffc4980 0x400e68b9:0x3ffc49a0 0x400e6929:0x3ffc49c0 0x400f3a3a:0x3ffc49e0 0x400eb05a:0x3ffc4a80 0x400e68b9:0x3ffc4ae0 0x400e68e6:0x3ffc4b00 0x400dc4c7:0x3ffc4b20 0x400dc761:0x3ffc4bd0 0x400d46da:0x3ffc4bf0`

larsmm, 2019-07-14:
It is not fixed. I still have this problem in fff2e193d064effe36a7d456050faa78fe6280a8. Please help.

sunblade, 2021-03-01:
I have similar issues with receiving data over UART. From time to time received data is corrupted (detected by checsum). CPU is running on default freq.

`uart = UART(2, rx=34, tx=32, baudrate=115200, bits=8, parity=None, stop=1, timeout=500, buffer_size=4096)`

Data source: RAW GPS receiver
Payload size: 500-2500bytes