loboris/MicroPython_ESP32_psRAM_LoBo #137: socket.recv() and socket.read() sometimes hang forever

chrisb2, 2018-05-01:
I have implemented Response.iter_content() in urequests so that I can deal with large responses without running out of memory. I based my code on the equivalent in [Python requests](https://github.com/requests/requests/blob/2de907ad778de270911acaffe93883f0e2729a4a/requests/models.py#L724). My new method is:
```
    def iter_content(self, chunk_size=1):
        def generate():
            while True:
                chunk = self.raw.read(chunk_size)
                if not chunk:
                    break
                yield chunk
            self._content_consumed = True

        if self._content_consumed:
            raise RuntimeError("response already consumed")
        elif chunk_size is not None and not isinstance(chunk_size, int):
            raise TypeError("chunk_size must be an int, it is instead a %s."
                % type(chunk_size))

        return generate()
```
I copied urequests and renamed to requests before adding iter_content() to requests, so that I could have both modules for testing.

My test program that executes this is:
```
import requests

for x in range(1,20):
    r = requests.get('http://jsonplaceholder.typicode.com/users')
    try:
        for chunk in r.iter_content(chunk_size=256):
            print(chunk.decode('UTF-8'))
    finally:
        r.close()
```
Running this with either recv() or read() in iter_content() I get the following results:
- Loboris Micropython for ESP32 - always hangs before 20 iterations is reached
- Micropython for ESP32 - always hangs before 20 iterations is reached
- Micropython for ESP8266 - never hangs

The hang always occurs when the request looks complete in terms of what is printed out. If I press cntl-C when a hang has occurred I get:
```
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "<stdin>", line 4, in <module>
  File "requests.py", line 36, in generate
KeyboardInterrupt:
```
where line 36 is 
`chunk = self.raw.read(chunk_size)`

regards,
Chris

chrisb2, 2018-05-02:
Also raised this issue in Official MicroPython: [socket.recv() and socket.read() sometimes hang forever](https://github.com/micropython/micropython/issues/3754)

chrisb2, 2018-05-03:
This has now been fixed in Official Micropython, see [6681eb8](https://github.com/micropython/micropython/commit/6681eb809a815a049ca069d522b9ad07548db535)

chrisb2, 2018-05-05:
I applied the changes from the official Micropython repo, and tested the result, its working fine now. I have attached a patch file of the changes, but could raise PR if you like?
[issue-137.patch.txt](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/files/1976251/issue-137.patch.txt)

loboris, 2018-05-13:
The changes to the **socket** module from the official Micropython repo are included in the latest update.

chrisb2, 2018-05-23:
Have deployed updated firmware and can confirm its working as expected with no hangs so far. Thank you

closed 2018-05-23 by chrisb2