loboris/MicroPython_ESP32_psRAM_LoBo #139: How to change application partition's size?

eminarcissus, 2018-05-03:
Hi,

I'm trying to compile a firmware which is larger than 1M, I found that partitions_mpy.csv is auto generated, how can I change the size of the partition itself?

This is the bootlog

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4776
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13260
entry 0x40078d74
E (703) esp_image: Image length 1877008 doesn't fit in partition length 1048576
E (704) boot: Factory app partition is not bootable
E (706) boot: No bootable app partitions in the partition table

loboris, 2018-05-03:
All MicroPython images are larger than 1MB.

If you did not changed something, BUILD.sh will create correct partition table (with or without -a option).
I'm using this build system for more than a year and never had a problem.

You may try to erase the flash.

You must flash the firmware with `./BUILD.sh flash` (optionally using the options for flash fs and application sizes), the partition table is created before flashing.

bdespatis, 2018-05-04:
Yes, I know what it could be. We had a similar issue, but the solution is simple.

If you flash with -a, you have to use the same size for flashfs (spiffs image). If you don't, you will write your internal file system at the wrong address and make your 2nd partition or factory partition unusable.

eminarcissus, 2018-05-07:
Thx very much for the help, yes, that solved the problem

closed 2018-05-07 by eminarcissus