loboris/MicroPython_ESP32_psRAM_LoBo #140: GSM Functional ??

diginfo, 2018-05-04:
I am using SIM5320 which has same AT Commands as the SIM800 only supports 3G.

I am able to communicate with the device using the UART directly, but unable to get any response using the gsm module.

I have enabled PPP && PAP Support as well as enabling the gsm module:

**Using UART to get Signal Strength (SIM Card not installed)**

```
>>> import machine as mc
>>> uart = mc.UART(1, tx=22, rx=21, timeout=1000,  buffer_size=1024)
>>> uart.write('AT+CSQ\r\n')
8
>>> uart.any()
21
>>> uart.readln()
'\r\n'
>>> uart.readln()
'+CSQ: 10,99\r\n'
>>> uart.readln()
'\r\n'
>>> uart.readln()
'OK\r\n'
>>> uart.readln()
>>>
```

**Using GSM Module to get Signal Strength (SIM Card not installed)**
```
>>> import gsm
>>> gsm.start(tx=22, rx=21, apn='internet.ht.hr')
True
>>> gsm.status()
(98, 'Not started')
>>> gsm.atcmd('AT+CSQ')
''
```

Not sure if I have done something wrong, or omitted something, but communication via UART seems OK ?

loboris, 2018-05-08:
It is recommended to use **gsm module** for accessing GSM. See the [Wiki](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/gsm).
You can also use it with uart access, but the PPPoS will not be available.
When using gsm module, it will **not** enter the **idle mode** (in which you can use at commands) if the SIM card is not inserted).

```python
import gsm
>>> gsm.start(tx=25, rx=26, apn='internet.ht.hr')
True
>>> gsm.status()
(98, 'Not started')
>>> gsm.status()
(89, 'Idle')
# When not connected (in idle mode), we can issue the at commands
>>> gsm.atcmd('ATI4',printable=True)
'..SIM800 R13.08....OK..'
>>> gsm.atcmd('AT+CSQ',printable=True)
'..+CSQ: 29,0....OK..'
>>> 
# Check and read SMS
>>> gsm.checkSMS(sort=gsm.SORT_NONE, status=gsm.SMS_ALL)
(1,)
>>> gsm.readSMS(1)
(1, 'REC READ', '+385992382166', '18/05/08,09:22:58+08', 1525771378, 0, 'Hi, GSM module on MicroPython')
>>> 
# Go online
>>> gsm.connect()
True
>>> gsm.status()
(1, 'Connected')
>>> 
# While online all network functions works the same way as with Internet connection via WiFi
>>> 
# When Internet connection is no longer needed, disconnect
>>> gsm.disconnect()
>>> gsm.status()
(89, 'Idle')
>>> 
```

kindmartin, 2021-03-30:
now these days nothing as charm/ easy as this gsm module for esp32. its 2021 and current micropython official 1.14 now do not have this beauty :-(