loboris/MicroPython_ESP32_psRAM_LoBo #141: MQTT client not connecting to mosquitto broker with wifi Access Point mode

mzdaniel, 2018-05-05:
While trying mqtt (using wifi in AP mode), creating an mqtt object immediatelly returned "Error starting client".  Kept investigating and here is the result of my tests:

At fisrt I tried with my default hbmqtt broker (which I found the mqtt client can't work properly as will bit is set).  I installed mosquitto broker and ensured mqtt worked using wifi Station mode.

mqtt with wifi in Station mode (works as expected):

```python
from network import WLAN, mqtt
from time import sleep

w = WLAN()
w.active(True)
w.connect('espnet', 'asdf')
sleep(2)

m = mqtt('esp_sensor', '192.168.1.147')
sleep(2)
m.start()

m.publish('temperature', '42')
```

mqtt with wifi in Access Point mode (doesn't work):

```python 
from network import WLAN, AP_IF, mqtt

w = WLAN(AP_IF)
w.config(essid='espnet')
w.active(True)

# After ensuring wifi remote access point is up (and binging its mosquitto server)
w.wifiactive()  # Returns   True
w.ifconfig()     # Returns   ('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')

m = mqtt('esp_sensor', '192.168.4.2')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Error starting client
```

loboris, 2018-05-08:
I have recently updated the mqtt module (now uses the new mqtt library) and it works well in Access Point mode.
The update will be commited today or tomorrow.

Example (with debug log level), tested with Android [**MQTT Broker App**](https://play.google.com/store/apps/details?id=server.com.mqtt&rdid=server.com.mqtt) running on my phone:
```python
>>> from network import WLAN, AP_IF, mqtt
D (2829502) [modnetwork]: Initializing TCP/IP
D (2829503) [modnetwork]: Initializing Event Loop
D (2829503) [modnetwork]: Event loop initialized
>>> w = WLAN(AP_IF)
>>> w.config(essid='espnet')
I (2867353) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2867353) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (2867371) [modnetwork]: WiFi Initialized
>>> w.active(True)
W (2886571) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
I (2886718) phy: phy_version: 386.0, 67c798f, Mar 14 2018, 16:34:06, 0, 2
D (2886730) [modnetwork]: WiFi Started, mode 2
True
>>> w.wifiactive()
True
>>> w.ifconfig()
('192.168.4.1', '255.255.255.0', '192.168.4.1', '0.0.0.0')

# === Connected my phone to espnet AP ===

>>> m = mqtt('esp_sensor', '192.168.4.2')
>>> m.start()
>>> D (3093338) TRANSPORT_TCP: [sock=8192],connecting to server IP:192.168.4.2,Port:1883...

>>> m
Mqtt[esp_sensor]
    (Server: 192.168.4.2:1883, Status: Connected
     Client ID: mpy_mqtt_client, Clean session=False, Keepalive=120s
     LWT(not set)
    )

>>> m.status()
(2, 'Connected')
>>> m.publish('temperature', '42')
True
```

mzdaniel, 2018-05-08:
This is great. Thank you so much Boris for such an awesome project!

mzdaniel, 2018-05-11:
I can confirm your the issue is resolved in  ESP32_LoBo_v3.2.15 and on top of that HBMQTT broker also works as expected too. 

Thank you again Boris!

closed 2018-05-11 by mzdaniel