loboris/MicroPython_ESP32_psRAM_LoBo #142: REPL in separate thread, main thread stack too low

kevinkk525, 2018-05-12:
If I run my program with "REPL started in separate thread" I get this issue:
(had it since this feature was introduced)

***ERROR*** A stack overflow in task mp_task has been detected.
abort() was called at PC 0x40096d68 on core 1

Backtrace: 0x40096b88:0x3ffc0540 0x40096d4f:0x3ffc0560 0x40096d68:0x3ffc0580 0x40092d7e:0x3ffc05a0 0x40094f90:0x3ffc05c0 0x40094f46:0x00000000

CPU halted.

bdespatis, 2018-05-12:
Are you using the OTA features ? Do you flash your internal file system with flashfs ? Have you tried the precompile firmware available for test ?

kevinkk525, 2018-05-12:
No, No (has nothing to do with the problem anyways).
Precompiled firmware does not have that feature active.

I reported this problem on the forum when the feature was first introduced and loboris said he could try to increase the stack for the thread.

bdespatis, 2018-05-12:
Ok. I tried this feature and it works each time without a problem. That is why I was checking if your partition and file system is corrupted. I even had this feature activated the first on a board previously configured that I tried to connect with my new repository (monitor). It asked me and now it starts in a new thread.

kevinkk525, 2018-05-12:
Maybe your program is smaller and has therefore no problem.
Running it in the repl thread works.

loboris, 2018-05-13:
The main task (mp_task) stack siize with enabled `Start REPL in separate thread` was increased in the latest update.
If you build with MicroPython loglevel set to debug, all stack/heap sizes and addresses wil be reported:
```
D (139) MicroPython: Entry
D (1142) MicroPython: SPIRAM: Enabled
D (1151) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (1200) MicroPython: Configure stack
D (1200) nvs: nvs_open_from_partition MPY_NVM 1
D (1201) nvs: nvs_get MPY_StackSize 4
D (1201) nvs: nvs_get_str_or_blob MpyTimeZone
D (1206) MicroPython: MPy stack: 0x3ffc041c - 0x3ffc1424 (4104)
D (1212) MicroPython: Configure heap
D (1215) nvs: nvs_get MPY_HeapSize 4
D (1219) MicroPython: MPy heap: 0x3f80001c - 0x3fb8005c
D (1224) MicroPython: Main task min stack: 656
I (1234) MicroPython: [=== MicroPython FreeRTOS task started (sp=3ffc1350) ===]

D (1238) littleflash: LittleFS on top of ESP32 wear leveling
D (1243) littleflash: Original_partition_size=2432KB, WL_size=2392KB

Internal FS (LittleFS ver 1.4): Mounted on partition 'internalfs' [size: 2490368; Flash address: 0x1A0000]
----------------
Filesystem size: 2449408 B
           Used: 2560 B
           Free: 2446848 B
----------------

FreeRTOS running on BOTH CORES, MicroPython task started on App Core (1).

 Reset reason: Power on reset
   REPL stack: 16384 bytes
     uPY heap: 3585920/6336/3579584 bytes (in SPIRAM using heap_caps_malloc)

>>> Starting REPL in thread <<<

MicroPython ESP32_LoBo_v3.2.15 - 2018-05-10 on ESP32 board with ESP32
Type "help()" for more information.
>>> 
```