loboris/MicroPython_ESP32_psRAM_LoBo #144: ssd1306_spi_example.py appears to have two errors

peterhinch, 2018-05-13:
In [these lines](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/b7f1de4a45a902ce6d2403b2530def9fabe44f52/MicroPython_BUILD/components/micropython/esp32/modules_examples/ssd1306_spi_example.py#L9) the MOSI and MISO pins seem to be initialised incorrectly: MOSI should be an output and MISO an input.

loboris, 2018-05-13:
You are right, I'll change it.

It actually has no effect on operation, as the SPI pins are initialized the right way by the driver.
The SPI object can also be created by giving the pin numbers instead of Pin objects:
`spi = machine.SPI(1,baudrate=1000000, sck=19, mosi=23, miso=25)
`

loboris, 2018-05-15:
Updated.

closed 2018-05-15 by loboris