loboris/MicroPython_ESP32_psRAM_LoBo #149: CPU Halted when open webpage on microWebserver

elektrobil, 2018-05-18:
Im using last firmware MicroPython ESP32_LoBo_v3.2.17. After last update I have a this error When urequest module and MicroWebserver module running same time its mean when client -server get and post then I try to open a webpage to server yhis error appear and cpu halted.
`assertion "ATB_GET_KIND(block) == AT_HEAD" failed: file "/home/cem/MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/components/micropython/py/gc.c", line 699, function: gc_realloc`

loboris, 2018-05-19:
Could you provide the Python source which causes the crash?

It looks like some Python object is accessed which was changed elsewhere (probably in another thread) ...

closed 2018-09-11 by elektrobil

givemeaname, 2018-12-16:
@elektrobil i expierence the same issue. would you mind to share your solution if you got one?