loboris/MicroPython_ESP32_psRAM_LoBo #150: utimeq update makes async mqtt lag a lot until watchdog

kevinkk525, 2018-05-19:
I posted the issue on the [forum](https://loboris.eu/forum/Thread-Issue-report-Network-lag-results-in-watchdog) but I have additional information now:

If you change the function time_less_than(**) back to the function before the update, all works fine again. (Although the changes to that function make sense to me but the new version does not work).

#define MODULO MICROPY_PY_UTIME_TICKS_PERIOD

//----------------------------------------------------------------------
STATIC bool time_less_than(struct qentry *item, struct qentry *parent) {
    mp_uint_t item_tm = item->time;
    mp_uint_t parent_tm = parent->time;
    mp_uint_t res = parent_tm - item_tm;
    if (res == 0) {
        // TODO: This actually should use the same "ring" logic
        // as for time, to avoid artifacts when id's overflow.
        return item->id < parent->id;
    }
    if ((mp_int_t)res < 0) {
        res += MODULO;
    }
    return res && res < (MODULO / 2);
}

loboris, 2018-05-20:
The method of sorting the queue items in utimeq is a little strange.
I'm testing the new **utimeq** module which provides the queue of `qentry` items ( `struct qentry {int64_t time; mp_uint_t id; mp_obj_t callback; mp_obj_t args;}` ) sorted in proper way.

A few new functions and options are added:
- `len()` returns tuple of current queue length and allocated length
- `dump([True])` is provided which prints the detailed que information. If `True` argument is given, all queue items are printed, if not, only the active items
- `peek(idx, res)` is provided, which peeks the queue item form the index `idx`
- `peektime([idx])` accepts optional argument to peek the time value from queue at index `idx`. If the argument is not given, peeks the last value (as before).
- time value in a range of `int64_t` are accepted and properly sorted. `Floats `can be accepted as time value and are rounded to `int64_t`
- overflaw on time value newer occurs as the time range is more then 290000 years with micro seconds value.

You can replace the existing `modutimeq.c` with [this one](http://loboris.eu/ESP32/modutimeq.c) and rebuild to test it.

Example:
```python
# --- Ctrl+E, paste code
from utimeq import utimeq

def test():
    print("Test")

arg = (123, "Test")

h=utimeq(15)
item = [0,0,0]

h.push(100, 22, 0)
h.push(4294967400, 20, 0)
h.push(4294967350, 20, 0)
h.push(1000, 22, 0)
h.push(4294967350, 20, 0)
h.push(950, 22, 0)
h.push(1000, 22, 0)
h.push(7, 22, 0)
h.push(2000, 22, 0)
h.push(0x7fffffffffffffff, 22, 0)
h.push(2000, 0xBFFFFFFF, 0xCFFFFFEE)
h.push(120, test, arg)

h.dump(True)

h.peek(9, item)
print(item)
# --- Ctrl+D, execute

 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0  9223372036854775807        22        0x2d         0x1
   1           4294967400        14        0x29         0x1
   2           4294967350        17        0x29         0x1
   3           4294967350        15        0x29         0x1
   4                 2000        23  0x3ffcc380  0x3ffcc3a0
   5                 2000        21        0x2d         0x1
   6                 1000        19        0x2d         0x1
   7                 1000        16        0x2d         0x1
   8                  950        18        0x2d         0x1
   9                  120        24  0x3ffcac00  0x3ffcac30
  10                  100        13        0x2d         0x1
  11                    7        20        0x2d         0x1
  12                    0         0         0x0         0x0  (empty)
  13                    0         0         0x0         0x0  (empty)
  14                    0         0         0x0         0x0  (empty)
-----------------------------------------------------------
[120, <function test at 0x3ffcac00>, (123, 'Test')]

# Some tests
>>> h.pop(item)
>>> item
[9223372036854775807, 22, 0]
>>> h.dump(True)
 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0           4294967400        14        0x29         0x1
   1           4294967350        17        0x29         0x1
   2           4294967350        15        0x29         0x1
   3                 2000        23  0x3ffcc380  0x3ffcc3a0
   4                 2000        21        0x2d         0x1
   5                 1000        19        0x2d         0x1
   6                 1000        16        0x2d         0x1
   7                  950        18        0x2d         0x1
   8                  120        24  0x3ffcac00  0x3ffcac30
   9                  100        13        0x2d         0x1
  10                    7        20        0x2d         0x1
  11                    0         0         0x0         0x0  (empty)
  12                    0         0         0x0         0x0  (empty)
  13                    0         0         0x0         0x0  (empty)
  14                    0         0         0x0         0x0  (empty)
-----------------------------------------------------------
>>> h.pop(item)
>>> h.pop(item)
>>> h.pop(item)
>>> h.dump()
 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0                 2000        23  0x3ffcc380  0x3ffcc3a0
   1                 2000        21        0x2d         0x1
   2                 1000        19        0x2d         0x1
   3                 1000        16        0x2d         0x1
   4                  950        18        0x2d         0x1
   5                  120        24  0x3ffcac00  0x3ffcac30
   6                  100        13        0x2d         0x1
   7                    7        20        0x2d         0x1
-----------------------------------------------------------
>>> h.push(1120, test, arg)
>>> h.dump()
 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0                 2000        23  0x3ffcc380  0x3ffcc3a0
   1                 2000        21        0x2d         0x1
   2                 1120        25  0x3ffcac00  0x3ffcac30
   3                 1000        19        0x2d         0x1
   4                 1000        16        0x2d         0x1
   5                  950        18        0x2d         0x1
   6                  120        24  0x3ffcac00  0x3ffcac30
   7                  100        13        0x2d         0x1
   8                    7        20        0x2d         0x1
-----------------------------------------------------------
>>> h.len()
(9, 15)
>>> h.peektime()
2000
>>> h.peektime(5)
950
>>> 
```

kevinkk525, 2018-05-20:
I tried your new utimeq module and it behaves the same way. 
I think peektime() works exactly reversed. It should return the lowest time value, not the highest. 
Also your utimeq pops the highest time value while it should pop the lowest time value so the sorting is incorrect.

In the original version it returns the lowest time value. And that makes sense as asyncio is waiting for the next coroutine which of course has the lowest time value.

loboris, 2018-05-20:
OK, I've reversed the sort order, now it should work as expected.
I will probably add the optional sort argument (with the default value set to ascending) to the object create function.
[modutimeq.c](http://loboris.eu/ESP32/modutimeq.c) is updated.

```python
# --- Ctrl+E, paste code
from utimeq import utimeq

def test():
    print("Test")

arg = (123, "Test")

h=utimeq(15)
item = [0,0,0]

h.push(100, 22, 0)
h.push(4294967400, 20, 0)
h.push(4294967350, 20, 0)
h.push(1000, 22, 0)
h.push(4294967350, 20, 0)
h.push(950, 22, 0)
h.push(1000, 22, 0)
h.push(7, 22, 0)
h.push(2000, 22, 0)
h.push(0x7fffffffffffffff, 22, 0)
h.push(2000, 0xBFFFFFFF, 0xCFFFFFEE)
h.push(120, test, arg)

h.dump(True)

h.peek(9, item)
print(item)
# --- Ctrl+D, execute

 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0                    7         7        0x2d         0x1
   1                  100         0        0x2d         0x1
   2                  120        11  0x3f816150  0x3f816180
   3                  950         5        0x2d         0x1
   4                 1000         3        0x2d         0x1
   5                 1000         6        0x2d         0x1
   6                 2000         8        0x2d         0x1
   7                 2000        10  0x3f816c40  0x3f816c60
   8           4294967350         2        0x29         0x1
   9           4294967350         4        0x29         0x1
  10           4294967400         1        0x29         0x1
  11  9223372036854775807         9        0x2d         0x1
  12                    0         0         0x0         0x0  (empty)
  13                    0         0         0x0         0x0  (empty)
  14                    0         0         0x0         0x0  (empty)
-----------------------------------------------------------
[4294967350, 20, 0]
>>> h.pop(item)
>>> item
[7, 22, 0]
>>> h.dump()
 Idx                 Time        ID    Callback         Arg
-----------------------------------------------------------
   0                  100         0        0x2d         0x1
   1                  120        11  0x3f816150  0x3f816180
   2                  950         5        0x2d         0x1
   3                 1000         3        0x2d         0x1
   4                 1000         6        0x2d         0x1
   5                 2000         8        0x2d         0x1
   6                 2000        10  0x3f816c40  0x3f816c60
   7           4294967350         2        0x29         0x1
   8           4294967350         4        0x29         0x1
   9           4294967400         1        0x29         0x1
  10  9223372036854775807         9        0x2d         0x1
-----------------------------------------------------------
>>> 
```

kevinkk525, 2018-05-20:
This works as expected. 
Thanks a lot for your work on this! I'll close this issue once the module is published in the next firmware release.

mzdaniel, 2018-06-05:
Thank you so much @loboris and @kevinkk525 for this!   

I am getting almost exactly the same output as Boris with MicroPython ESP32_LoBo_v3.2.18 - 2018-06-04 on ESP32 board with ESP32 enabling uasyncio and picoweb.

I made a few comments of the findings in https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/148 for tiny hello world.  

In general, wonder if it would be useful to have some 'standard' tests as part of this repo

kevinkk525, 2018-06-05:
oh i have no part in this port, I'm just reporting bugs I encountering while developing, no need to thank me ;)

In #148 you probably encounter the utimeq problem, at least it sounds like it. Try the file linked here.

mzdaniel, 2018-06-06:
@kevinkk525: sorry if I confused you, I meant ESP32_LoBo_v3.2.18 is now reproducing the output as in https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/150#issuecomment-390485258 in the tests and picoweb is working for me with simple and light usecases. How are you testing umqtt?  Maybe I can help.

kevinkk525, 2018-06-06:
my bad, did not check if a new commit was published. Thanks loboris.

Well the comments in #150 are pretty much our test cases :D Thanks for your offer but my guess is, that utimeq is tested well enough now. Or at least the features we need for uasyncio are, feel free to test every function again. One can never be certain there is no bug but only prove that there is one.

mzdaniel, 2018-06-06:
This uasync issue on this port could have taken months if even resolved without your effort and tests. Thank you again @kevinkk525 for finding what appears to be the root cause.
It also looks to me we can put this issue to rest now. I am planning to do more extensive tests in the following weeks in both picoweb and umqtt.

closed 2018-06-06 by kevinkk525