loboris/MicroPython_ESP32_psRAM_LoBo #151: machine.disable_irq() - Guru meditation error

df6dbf, 2018-05-20:
Hello Folks,
got a serious error while try to debounce a switch triggered an irq.

MicroPython ESP32_LoBo_v3.2.15 - 2018-05-10 on ESP32 board with ESP32
Type "help()" for more information.
>>> machine.disable_irq()
393504
>>> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008e2b9  PS      : 0x00060034  A0      : 0x8008eb7f  A1      : 0x3ffb0fa0  
A2      : 0x3ffb2470  A3      : 0x00000000  A4      : 0x00060c20  A5      : 0x00000001  
A6      : 0x00060c23  A7      : 0x00060023  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00060021  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x400d3e3a  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008e2b9

Backtrace: 0x4008e2b9:0x3ffb0fa0 0x4008eb7c:0x3ffb0fc0 0x4008f953:0x3ffb0fe0 0x400905b9:0x3ffb1000 0x400828fa:0x3ffb1010 0x400d3e37:0x00000000

Core 1 register dump:
PC      : 0x40091720  PS      : 0x00060b34  A0      : 0x8008ee92  A1      : 0x3ffc3d10  
A2      : 0x3ffbd1f4  A3      : 0x3ffb3544  A4      : 0x00060b23  A5      : 0x00000001  
A6      : 0x00060b23  A7      : 0x3f415730  A8      : 0x3ffb3544  A9      : 0x3ffb3544  
A10     : 0x00000014  A11     : 0x00000014  A12     : 0x00060b23  A13     : 0x00000001  
A14     : 0x00060b23  A15     : 0x3ffc3c80  SAR     : 0x0000000a  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x40091720:0x3ffc3d10 0x4008ee8f:0x3ffc3d30 0x400902fb:0x3ffc3d50 0x400d4f39:0x3ffc3d90 0x400de68c:0x3ffc3dc0 0x400dea03:0x3ffc3de0 0x400d4930:0x3ffc3e20

CPU halted.

Kind regard, Holger

loboris, 2018-05-20:
The interrupts must never be disable for more than the value set in **menuconfig** (default 300 ms):
`→ Component config → ESP32-specific → Interrupt watchdog timeout (ms)`

Using `machine.disable_irq()` is not recommended on ESP32 if you cannot guarantee that the disabled time will be short enough.

Debouncing pin interrupts will be added to **machine.Pin** module soon.

eflukx, 2018-07-23:
Ok, fair enough you may not disable global interrupts for longer periods. But any ideas on what causes the hard fault? (Mostly out of interest)

yeyeto2788, 2018-09-17:
Hello guys! I actually got the same error but not even using (As long as I know)
 the `machine.disable_irq()` 

This is the error I got:
```python
>>> from console import Screen
>>> oled = Screen()
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x401089d6  PS      : 0x00060d30  A0      : 0x800ef811  A1      : 0x3ffc5c80
A2      : 0x400f4ca2  A3      : 0x3f416120  A4      : 0x3ffc5d90  A5      : 0x00000405
A6      : 0xffffffff  A7      : 0x000000a0  A8      : 0x801089d1  A9      : 0x3ffc5c50
A10     : 0x00000009  A11     : 0x00000009  A12     : 0x00000001  A13     : 0x00000014
A14     : 0x3ffc5c80  A15     : 0x3ffc5d6c  SAR     : 0x0000000c  EXCCAUSE: 0x00000009
EXCVADDR: 0x400f4ca2  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x401089d6:0x3ffc5c80 0x400ef80e:0x3ffc5d90 0x400eafc1:0x3ffc5dc0 0x400eb031:0x3ffc5de0 0x400f8441:0x3ffc5e00 0x400ef89e:0x3ffc5ea0 0x400eafc1:0x3ffc5ed0 0x400eb031:0x3ffc5ef0 0x400f458f:0x3ffc5f10 0x400f3c03:0x3ffc5f60 0x400eafc1:0x3ffc5f80 0x400f83b5:0x3ffc5fa0 0x400ef89e:0x3ffc6040 0x400eafc1:0x3ffc60a0 0x400eafee:0x3ffc60c0 0x400ddd03:0x3ffc60e0 0x400ddf81:0x3ffc6190 0x400d5578:0x3ffc61d0

CPU halted.
```

It seems that the error is coming from creating a module to wrap display and keypad functionality into one module as I did on [my project](https://github.com/yeyeto2788/MicroPythonScripts/tree/master/MicroScriptConsole) 

This is the early stage of the module:

```python
import machine 
from display import TFT
import time
import math

class Screen(TFT):

    def __init__(self):
        self.init(self.ST7789, rst_pin=17, backl_pin=0, miso=19, mosi=23, clk=18, cs=5, dc=16, speed=32000000, splash=False)
        self.clear()
		self.(self.CENTER, 2, "tft.LANDSCAPE_FLIP", self.GREEN, transparent=True)
```

I also tried to handle the exception but no success on it:

```python
>>> from console import Screen
>>> try:
...     oled = Screen()
... except Exception:
...     print(str(Exception))
...
...
...
Guru Meditation Error: Core  1 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 1 register dump:
PC      : 0x401089d6  PS      : 0x00060d30  A0      : 0x800ef811  A1      : 0x3ffc5c60
A2      : 0x400f4ca2  A3      : 0x3f416120  A4      : 0x3ffc5d70  A5      : 0x00000405
A6      : 0xffffffff  A7      : 0x000000a0  A8      : 0x801089d1  A9      : 0x3ffc5c30
A10     : 0x00000009  A11     : 0x00000009  A12     : 0x00000001  A13     : 0x00000014
A14     : 0x3ffc5c60  A15     : 0x3ffc5d4c  SAR     : 0x0000000c  EXCCAUSE: 0x00000009
EXCVADDR: 0x400f4ca2  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace: 0x401089d6:0x3ffc5c60 0x400ef80e:0x3ffc5d70 0x400eafc1:0x3ffc5da0 0x400eb031:0x3ffc5dc0 0x400f8441:0x3ffc5de0 0x400ef89e:0x3ffc5e80 0x400eafc1:0x3ffc5eb0 0x400eb031:0x3ffc5ed0 0x400f458f:0x3ffc5ef0 0x400f3c03:0x3ffc5f40 0x400eafc1:0x3ffc5f60 0x400f83b5:0x3ffc5f80 0x400ef89e:0x3ffc6020 0x400eafc1:0x3ffc60a0 0x400eafee:0x3ffc60c0 0x400ddd03:0x3ffc60e0 0x400ddf81:0x3ffc6190 0x400d5578:0x3ffc61d0

CPU halted.
```

Any possible solution for this?

janoist1, 2019-06-26:
@loboris Thanks for the clarification on this. Can you tell me the status of the debounce feature? If not yet implemented, what's the best workaround?