loboris/MicroPython_ESP32_psRAM_LoBo #153: Error in the build process of the lates firmware

fabianWhoknows, 2018-05-22:
@loboris 
I wanted to build the latest version of your firmware and got the following error:
```
Building partitions from ../MicroPython_ESP32_psRAM_LoBo/MicroPython_BUILD/partitions_mpy.csv...
Traceback (most recent call last):
  File "../MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/partition_table/gen_esp32part.py", line 423, in <module>
    main()
  File "../MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/partition_table/gen_esp32part.py", line 395, in main
    table_size = table.flash_size()
  File "../MicroPython_ESP32_psRAM_LoBo/Tools/esp-idf/components/partition_table/gen_esp32part.py", line 122, in flash_size
    last = sorted(self, reverse=True)[0]
TypeError: '<' not supported between instances of 'PartitionDefinition' and 'PartitionDefinition'

```

I checked the line 122 in gen_esp32part.py but I don't see anything related to a '<' character there.. Of core I tried the build process with a fresh and untouched copy of your repository..

loboris, 2018-05-22:
It is sometimes useful to search the closed issues.
Look at [this one](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/147).

fabianWhoknows, 2018-05-22:
I'm sorry! Just found the answer on your forum as well (https://loboris.eu/forum/Thread-Problem-compiling-with-latest-build) Mea Culpa!!

closed 2018-05-22 by fabianWhoknows