loboris/MicroPython_ESP32_psRAM_LoBo #154: telnet doesn't work for me

schroepfer, 2018-05-23:
Hello,
i can't connect to telnet.
The following i have done:
**import network
import machine
wlan = network.WLAN(network.STA_IF); wlan.active(True)
wlan.scan()                             # Scan for available access points
wlan.connect("FRITZ2", "IhJSuwiK")      # Connect to an AP
wlan.isconnected()                      # Check for successful connection
rtc = machine.RTC()
rtc.ntp_sync(server="pool.ntp.org")
rtc.now()
network.telnet.start() 
print(wlan.ifconfig())
print(network.telnet.status())**

_>>> print(wlan.ifconfig())
('192.168.8.101', '255.255.255.0', '192.168.8.1', '192.168.8.1')
>>> print(network.telnet.status())
(1, 'Starting', ('192.168.8.101',))_

It looks fine, i tested telnet with several programs, but no connection. (Port 23 is right?)

Regards
Juergen

loboris, 2018-05-24:
When telnet server is ready for connection, `network.telnet.status()` should return `(2, 'Listen', ('ip_address',))`
`(1, 'Starting', ('192.168.8.101',))` indicates the the server cannot get the socket...
I've never had such an issue, I'll try to investigate it.

flashypepo, 2018-08-04:
I've same  problem (telnet client times out) despite the telnet.status `(2, 'Listen', 'ip-address')`. This applies for ESP32_LoBo_v3.2.20.

However, when I'm using the same code in ESP32_LoBo_v3.2.17, telnet client does connect.

Code snippet:
`def _startServices(self):
        """startServices() - start Bonjour name, telnet and ftp services"""
        from network import mDNS, ftp, telnet
        mdns = mDNS()
        mdns.start('myTTGO', 'TTGO-ESP32-T4')
        ftp.start(user='****', password='***')
        telnet.start(user='****', password='****')
        print('FTP status:', ftp.status())
        print('Telnet status:', telnet.status())`

Hope this will help in finding the problem.
(sorry for lousy format of code, I can't get it correct).
Peter

Note: Thanks for all the effort in creating this wonderful Micropython variant.