loboris/MicroPython_ESP32_psRAM_LoBo #155: WebREPL Support

amirgon, 2018-05-23:
Micropython now supports WebREPL also for ESP32 port.
See: https://github.com/micropython/micropython-esp32/issues/107

Any plans to support this on MicroPython_ESP32_psRAM_LoBo?

loboris, 2018-05-24:
Not for now. I don't think it is necessary, as you have other (better) means to connect to the REPL ower WiFi (Telnet & FTP).
If you want to connect from browser, you can use Chrome Telnet app [DuckClient](https://chrome.google.com/webstore/detail/duckclient/ghdjfcifcempcjefhonfpafmpelnmkho), it works quite well.

[image]

elf55, 2018-05-25:
If you want to use a browser to access, try the Jupyter notebook kernel at https://github.com/adafruit/jupyter_micropython_kernel.  It's set up for a /dev/tty interface, but you can use a telnet interface if you are clever.

closed 2018-05-25 by amirgon